BradenM/micropy-cli #38: <SYMLINKD> is missing when make a git Clone

fpe, 2019-07-30:
Hi
When I get an GIT Clone of my project on new PC, the <SYMLINKD> for -esp-micropython-1.11.0 and micropython is not created.

BradenM, 2019-07-30:
Hi @fpe,

Strange. I can't seem to reproduce it on my end.
If it exists, try removing the copied projects `.micropy` folder and re-running `micropy`.

Do you mind providing me with a copy of your `$HOME/.micropy/micropy.log` file? 
Also, if its public, mind providing me a link to the repo your using?

Going to need a bit more information to get to the bottom of this one :)

fpe, 2019-07-31:
First, thank you for your response.
I have now deleted the .micropy folder and run "micropy install" in cmd as admin, and it worked and everything is in place now. After looking at it a little close, I can see that on it I am called "fpe" and on the other I am called "Flemm". I enclose micropy.log as well as the link to github.

https://flemmingpetersen.visualstudio.com/DefaultCollection/AutomaticPlantBox/_git/AutomaticPlantBox

Thanks for the help

Best regrades
Flemming Petersen

BradenM, 2019-07-31:
Glad to see you got it working.

For the record, I do not have permission to view the repo you linked, but since your issue has been resolved all is well. 

I will add a more descriptive error for Windows users when they try to execute micropy without the proper permissions to avoid any future confusion. Thanks!

closed 2019-07-31 by BradenM