BradenM/micropy-cli #40: BUG: TypeError: Unknown color 'bright_green'

patvdleer, 2019-08-21:
```
$ micropy --version
micropy, version 2.0.1
```

```
$ micropy init
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/click/termui.py", line 372, in style
    bits.append('\033[%dm' % (_ansi_colors.index(fg) + 30))
ValueError: tuple.index(x): x not in tuple

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/micropy", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/micropy/cli.py", line 63, in init
    mp = MicroPy()
  File "/usr/local/lib/python3.7/dist-packages/micropy/main.py", line 26, in __init__
    self.setup()
  File "/usr/local/lib/python3.7/dist-packages/micropy/main.py", line 30, in setup
    self.log.debug("\n---- MicropyCLI Session ----")
  File "/usr/local/lib/python3.7/dist-packages/micropy/logger.py", line 274, in debug
    return self.debug(msg, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/micropy/logger.py", line 275, in debug
    self.echo(msg, log="debug")
  File "/usr/local/lib/python3.7/dist-packages/micropy/logger.py", line 164, in echo
    service_title = self.get_service(fg=title_color, bold=title_bold)
  File "/usr/local/lib/python3.7/dist-packages/micropy/logger.py", line 135, in get_service
    f"{self.service_name}", fg=color, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/termui.py", line 374, in style
    raise TypeError('Unknown color %r' % fg)
TypeError: Unknown color 'bright_green'
```

patvdleer, 2019-08-21:
You need to add click to the requirements, `click >=7`

BradenM, 2019-08-21:
Hi @patvdleer, thanks for the report! 

Just tested this out and found that you are correct! I will update the click requirement soon. Thanks!

commit 6f75e191 (2019-08-21): fix(dep): Require appropriate Click version

Updated click requirement to be version 7 or above

Fixes #40
closed 2019-08-21 by BradenM