BradenM/micropy-cli #43: Add support for Lego EV3 MicroPython

vbolshakov, 2019-08-29:
Add support for Lego EV3 MicroPython. https://education.lego.com/en-us/support/mindstorms-ev3/python-for-ev3

BradenM, 2019-08-29:
Hi, thanks for the idea. 

I will look further into it, but as I do not own an EV3 brick myself, I am not sure if I will be able to generate the stubs for it. Have you tried executing `micropy stubs create <EV3_PORT>`?

If the stubs generate successfully, you could either provide them to me or create a PR on [micropy-stubs](https://github.com/BradenM/micropy-stubs) to add them for everyone. 

Let me know if you're able to make any progress with this. Thanks.

vbolshakov, 2019-08-29:
There is no micropython repl on uart, but there is ssh access to ev3. Just need to understand what scripr i need to run on ev3 to generate stubs.

vbolshakov, 2019-08-29:
Can you add option to generate stubber as file?

BradenM, 2019-08-29:
Interesting. 

So micropy-cli uses [micropython-stubber](https://github.com/Josverl/micropython-stubber) for its stub generation. Essentially, you need to do the following:

1. Clone `micropython-stubber` and open `createstubs.py` in your editor.
2. Update the [self.modules list](https://github.com/Josverl/micropython-stubber/blob/030653d29abe9c583d706689f5d62d7f36f7d565/createstubs.py#L53) with the custom modules that are on the EV3
3. Upload and execute `createstubs.py`
4. Download the generated `stubs` directory (if it succeeds on the EV3)

If you run into any memory related errors, you can use [my fork](https://github.com/BradenM/micropython-stubber/tree/feat/process) of the repo, make your module edits, then execute `python process.py minify`. This will created a smaller `minified.py` that uses less memory.

For finding the EV3 modules you can reference their documentation and try executing `help('modules')` in repl (on most forks, this will print out a list of modules).

If you run into any trouble, I am happy to help.

vbolshakov, 2019-08-29:
There custom Lego build of micropython - pybricks-micropython (MicroPython v1.9.4 on 2019-03-26; linux version) with differences in 'os' module, so i change createstubs in several places with os and firmware version parsing. It is generating stubs for everything but custom modules. And i don't know why...
If you interested, i can provide remote access to ev3.

BradenM, 2019-08-29:
Sure, I'd love to take a look at it!

vbolshakov, 2019-08-30:
What is the best way to give you access credential?

Josverl, 2019-08-30:
Hi,

I think that what is happening is that there is no way for a .py script to find out which modules there are on the firmware.

the only way is to issue the command : `help('modules')` , however the output of that cannot be redirected , exept on specific hardware.

so the apporach that I have uses in micro-stubber is : 
- DO that once manually , per firmware 
- create an array of all possible combinations
- just try them all 

so : you need to add the lego specific ones .
to do that in   `createstubs.py` add a (few) lines with the LEGO custom modules :
``` python

    stubber.add_modules(['pybricks']) # subs for LEGO Pybricks
    stubber.create_all_stubs()
```

run createsubs.py 

retrieve (upload..) the stubs from the brick to your PC 
[image]

( and please share  or send a PR to https://github.com/josverl/micropython-stubber including any license ( MIT?) that is relevant for LEGO )

Josverl, 2019-08-30:
also if you have a list of LEGO specific MicroPython modules , i can just add them to the list of modules to try for micros-stubber.

vbolshakov, 2019-08-30:
help('modules') show modules without any problems:
`robot@ev3dev:~$ pybricks-micropython
MicroPython v1.9.4 on 2019-03-26; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> help('modules')
__main__          parameters_c      pybricks/uev3dev/messaging          umachine
_thread           pybricks/__init__ pybricks/uev3dev/sound              uos
array             pybricks/display  pybricks/uev3dev/util               urandom
boot              pybricks/ev3brick sys               ure
btree             pybricks/ev3devices                 termios           uselect
builtins          pybricks/ev3devio tools             usocket
cmath             pybricks/parameters                 ubinascii         ussl
ev3brick_c        pybricks/robotics ucollections      ustruct
ev3devices_c      pybricks/speaker  ucryptolib        utime
ffi               pybricks/tools    uctypes           utimeq
framebuf          pybricks/uev3dev/__init__           uerrno            uzlib
gc                pybricks/uev3dev/_alsa              uhashlib          websocket
math              pybricks/uev3dev/_wand              uheapq
micropython       pybricks/uev3dev/display            uio
mmap              pybricks/uev3dev/i2c                ujson
Plus any modules on the filesystem
`
But when i add stubber.add_modules(['pybricks/ev3brick'])

I got only this as result 
"""
Module: 'pybricks.ev3brick' on 1
"""
# MCU: posix
# Stubber: 1.2.0
display = None
ev3brick = None
parameters = None
speaker = None
tools = None
uev3dev = None

Josverl, 2019-08-30:
Okay, that is helpful 

I suspect that has something to do with nested modules ( module/submodule and even module/sub/leaf  ) 
( that is somthing I have noticed ( with M5Stack) before , but have not untangled earlier 

have you tried `stubber.add_modules(['pybricks'])` 
behaviour will depend on the implementation. It looks like LEGO has used frozen py-modules so we only can guess

Josverl, 2019-08-30:
also likely relevant : a manual stubbing exercise : https://github.com/drewwhis/pybricks-stubs

vbolshakov, 2019-08-30:
Already use it. But automation with micropy-cli will be better.

vbolshakov, 2019-08-30:
Checked both:
    stubber.add_modules(['pybricks']) # stubs for LEGO Pybricks
    stubber.add_modules(['pybricks/ev3brick']) # stubs for LEGO Pybricks

BradenM, 2019-08-30:
@Josverl, thanks for your help. I don't really know much about the EV3 products. 

@vbolshakov, agreed. The automated stub creation is relatively primative and could definitely be expanded on. 

When I get to a desktop later today I will look at this in more detail.

BradenM, 2019-08-31:
@vbolshakov, I suppose I could add [pybricks-stubs](https://github.com/drewwhis/pybricks-stubs) to micropy-stubs until Josverl/micropython-stubber#7 is solved. Sound good for now?

vbolshakov, 2019-08-31:
Yes.

BradenM, 2019-08-31:
Okay, went ahead and added what I could. Since it's only the extra modules from `pybricks` I would suggest adding another package along side it.

This seemed to work pretty well for me while I was testing it:
```sh
micropy stubs add ev3-pybricks-0.0.1
micropy stubs add esp32-micropython-1.9.4 # Someone mentioned PyBricks was based on 1.9.4?
```

```json
{
    "name": "NewProject",
    "stubs": {
        "esp32-micropython-1.9.4": "1.2.0",
        "ev3-pybricks-0.0.1": "1.2.0"
    },
    "config": {
        "vscode": true,
        "pylint": true
    },
    "packages": {},
    "dev-packages": {}
}
```

Josverl, 2019-09-01:
@vbolshakov 
Please take a look at : https://github.com/Josverl/micropython-stubber/issues/7#issuecomment-526954206

BradenM, 2019-09-05:
Closed via BradenM/micropy-stubs#3 and Josverl/micropython-stubber#7

Also, as I edited [here](https://github.com/BradenM/micropy-cli/issues/43#issuecomment-526656278), did not mean to bash on @Josverl 's work (my own, rather) haha. `micropython-stubber` is the reason we are even having this conversation.

closed 2019-09-05 by BradenM