BradenM/micropy-cli #44: Create Directory Junction instead making symbolic links on Windows

vbolshakov, 2019-08-29:
Junction (mklink /J) can be made without administrative privileges on Windows.

BradenM, 2019-08-29:
Thanks! I will definitely take a look at this when I get a chance today.

Josverl, 2019-08-30:
mklink

For windows 10  users who have Developer Mode enabled, the mklink command will now successfully create a symlink even f the user is not running as an administrator.

You can go to the For developers page at any time to enable or disable Developer Mode: simply enter "for developers" into the Cortana search box in the taskbar.

vbolshakov, 2019-08-30:
Hard link and Directory Junction differ. Directory Junction created without any problems

vbolshakov, 2019-08-30:
X:\!(2019-08-30 17-50)>mklink /H B A
Access is denied.

X:\!(2019-08-30 17-50)>mklink /J B A
Junction created for B <<===>> A

vbolshakov, 2019-08-30:
Also, when using Win API there is no problems to create Hard Link - i use Far (File and achieve manager) and able to create both without admin privileges.

BradenM, 2019-08-30:
When I get to my desktop today, I'm thinking about implementing a fallback on windows like this:

```py
import ctypes

kdll = ctypes.windll.LoadLibrary("kernel32.dll")

kdll.CreateSymbolicLinkW(UR"<project_dir>\.micropy\some-stub-link", UR"<micropy_folder>\some-stub", 1)
```

Thoughts?

vbolshakov, 2019-08-30:
Not work. Create Symlink only with administrative privileges.

vbolshakov, 2019-08-30:
Ok. There is no easy way... Pywin32 ad requirement or https://github.com/juntalis/ntfslink-python that just copy part of Pywin32.

BradenM, 2019-08-30:
Oh, for some reason I thought the `CreateSymbolicLinkW` method used a junction. 

I will take a look at `ntfslink-python`, thanks.

PR #45 (2019-09-01): fix(windows): Support User Level Directory Linking
Allows `micropy-cli` to be executed in a shell without admin privs by falling back on `mklink` when attempting to create a link on windows systems.

As discussed in #44 

Closes #44
closed 2019-09-01 by BradenM