BradenM/micropy-cli #47: Not work on Pyboard (and clones)

vbolshakov, 2019-09-05:
micropy-cli version 2.1.0 not work on Pyboard (STM32F405RGT6 MCU) and it's clones.
stubs generation end with errors. On ESP32 work without problems.
```
micropy stubs create COM8
...
MicroPy  Pyboard  Stub module: writer
MicroPy  Pyboard  Stub module: ymodem
MicroPy  Pyboard  Stub module: zlib
MicroPy  ✔ Done!
MicroPy  Copying stubs...
Source directory /pyboard/stubs does not exist.
Traceback (most recent call last):
  File "d:\applications\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\applications\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Applications\Python37\Scripts\micropy.EXE\__main__.py", line 9, in <module>
  File "d:\applications\python37\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "d:\applications\python37\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "d:\applications\python37\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\applications\python37\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\applications\python37\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\applications\python37\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "d:\applications\python37\lib\site-packages\micropy\cli.py", line 224, in create
    return mp.create_stubs(port, verbose=verbose)
  File "d:\applications\python37\lib\site-packages\micropy\main.py", line 83, in create_stubs
    stub_path = next(out_dir.iterdir())
StopIteration
```
After physically device disconnect-connect there is stub files visible on filesystem.  
Main difference of this boards is internal fs exposed as USB mass storage device. Disabling USB MSD not solve this problem.

BradenM, 2019-09-05:
Thanks for the report! 

Does the Pyboard use `/flash` as it's root directory? If so, I most likely forgot to implement a check to see if `/flash` exists before attempting to retrieve the stubs.

BradenM, 2019-09-05:
Also, could you share the log file via [pastebin](https://pastebin.com/)? It should be located under `$HOME\.micropy\micropy.log`

Thanks.

vbolshakov, 2019-09-05:
There is no "/flash" on pyboard, just "/". Are there "/flash" on ESP32? os.listdir() show only "/" content with main.py and boot.py.

https://pastebin.com/LiJVSyer

vbolshakov, 2019-09-05:
Stubs exists on flash if i connect to pyboard after micro-cli.

```
>>> os.listdir()
['boot.py', 'main.py', 'pybcdc.inf', 'README.txt', 'System Volume Information', 'stubs']
>>> os.listdir('stubs')
['pyboard_v1_11_on_2019']
>>> os.listdir('stubs/pyboard_v1_11_on_2019')
['uasyncio', 'umqtt', 'urllib', '_thread.py', 'array.py', 'binascii.py', 'builtins.py', 'cmath.py', 'collections.py', 'dht.py', 'errno.py', 'framebuf.py', 'gc.py', 'hashlib.py', 'heapq.py', 'io.py', 'json.py', 'machine.py', 'math.py', 'micropython.py', 'network.py', 'onewire.py', 'os.py', 'random.py', 're.py', 'select.py', 'socket.py', 'struct.py', 'sys.py', 'time.py', 'ubinascii.py', 'ucollections.py', 'uctypes.py', 'uerrno.py', 'uhashlib.py', 'uheapq.py', 'uio.py', 'ujson.py', 'uos.py', 'urandom.py', 'ure.py', 'uselect.py', 'usocket.py', 'ustruct.py', 'utime.py', 'utimeq.py', 'uzlib.py', 'zlib.py', 'modules.json']

```

BradenM, 2019-09-05:
Hmm. For some reason rshell isn't able to access the stub directory on the device... I will check this out soon, but for now I would make sure you have the latest rshell version via: `pip install rshell --upgrade`

Also, I re-released the pybricks stubs yesterday with the new generated stubs. You can install them via: `micropy stubs add ev3-pybricks-1.0.0` if you want to check them out.

vbolshakov, 2019-09-05:
```
Requirement already up-to-date: rshell in d:\applications\python37\lib\site-packages (0.0.26)
Requirement already satisfied, skipping upgrade: pyserial in d:\applications\python37\lib\site-packages (from rshell) (3.4)
Requirement already satisfied, skipping upgrade: pyudev>=0.16 in d:\applications\python37\lib\site-packages (from rshell) (0.21.0)
Requirement already satisfied, skipping upgrade: pyreadline in d:\applications\python37\lib\site-packages (from rshell) (2.1)
Requirement already satisfied, skipping upgrade: six in d:\applications\python37\lib\site-packages (from pyudev>=0.16->rshell) (1.12.0)
```
ev3-pybricks-1.0.0 already in use. Thank you again.

jmannau, 2019-09-30:
I have just experienced the same issue.

Here is my setup:
 - MacOS 10.14
 - VS Code
 - Pycom FiPy - v1.20rc13 FW
 - USB Serial Connection

Here is the verbose log 

```
$ micropy stubs create /dev/tty.usbmodemPycd040b1 --verbose

MicroPy  Connecting to Pyboard @ /dev/tty.usbmodemPycd040b1
Connecting to /dev/tty.usbmodemPycd040b1 (buffer-size 512)...
Trying to connect to REPL  connected
Testing if sys.stdin.buffer exists ... N
Retrieving root directories ... /flash/
Setting time ... Sep 30, 2019 20:40:56
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
MicroPy  ✔ Connected!
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Stub module: uasyncio/core
MicroPy  Pyboard  Stub module: umqtt/robust
MicroPy  Pyboard  Stub module: umqtt/simple
MicroPy  Pyboard  Stub module: urllib/urequest
MicroPy  Pyboard  Stub module: _thread
MicroPy  Pyboard  Stub module: ak8963
MicroPy  Pyboard  Stub module: apa102
MicroPy  Pyboard  Stub module: apa106
MicroPy  Pyboard  Stub module: array
MicroPy  Pyboard  Stub module: binascii
MicroPy  Pyboard  Stub module: btree
MicroPy  Pyboard  Stub module: builtins
MicroPy  Pyboard  Stub module: upip
MicroPy  Pyboard  Stub module: cmath
MicroPy  Pyboard  Stub module: collections
MicroPy  Pyboard  Stub module: curl
MicroPy  Pyboard  Stub module: dht
MicroPy  Pyboard  Stub module: display
MicroPy  Pyboard  Stub module: ds18x20
MicroPy  Pyboard  Stub module: errno
MicroPy  Pyboard  Stub module: esp
MicroPy  Pyboard  Stub module: esp32
MicroPy  Pyboard  Stub module: example_pub_button
MicroPy  Pyboard  Stub module: example_sub_led
MicroPy  Pyboard  Stub module: flashbdev
MicroPy  Pyboard  Stub module: framebuf
MicroPy  Pyboard  Stub module: freesans20
MicroPy  Pyboard  Stub module: functools
MicroPy  Pyboard  Stub module: gc
MicroPy  Pyboard  Stub module: gsm
MicroPy  Pyboard  Stub module: hashlib
MicroPy  Pyboard  Stub module: heapq
MicroPy  Pyboard  Stub module: inisetup
MicroPy  Pyboard  Stub module: io
MicroPy  Pyboard  Stub module: json
MicroPy  Pyboard  Stub module: logging
MicroPy  Pyboard  Stub module: lwip
MicroPy  Pyboard  Stub module: machine
MicroPy  Pyboard  Stub module: math
MicroPy  Pyboard  Stub module: microWebSocket
MicroPy  Pyboard  Stub module: microWebSrv
MicroPy  Pyboard  Stub module: microWebTemplate
MicroPy  Pyboard  Stub module: micropython
MicroPy  Pyboard  Stub module: mpu6500
MicroPy  Pyboard  Stub module: mpu9250
MicroPy  Pyboard  Stub module: neopixel
MicroPy  Pyboard  Stub module: network
[... 59 lines cut ...]
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/micropy/cli.py", line 233, in create
    return mp.create_stubs(port, verbose=verbose)
  File "/usr/local/lib/python3.7/site-packages/micropy/main.py", line 83, in create_stubs
    stub_path = next(out_dir.iterdir())
StopIteration
```

When I log into the device, the stubs are generated.
```
>>> os.listdir()
['boot.py', 'cert', 'lib', 'main.py', 'stubs', 'sys']
>>> os.listdir('/flash/stubs')
['FiPy_v1_9_4']
>>> os.listdir('/flash/stubs/FiPy_v1_9_4')
['_thread.py', 'array.py', 'binascii.py', 'builtins.py', 'cmath.py', 'errno.py', 'framebuf.py', 'gc.py', 'hashlib.py', 'json.py', 'machine.py', 'math.py', 'micropython.py', 'modules.json', 'network.py', 'os.py', 're.py', 'select.py', 'socket.py', 'ssl.py', 'struct.py', 'sys.py', 'time.py', 'uasyncio', 'ubinascii.py', 'ucollections.py', 'uctypes.py', 'uerrno.py', 'uhashlib.py', 'uio.py', 'ujson.py', 'umqtt', 'uos.py', 'ure.py', 'urllib', 'uselect.py', 'usocket.py', 'ussl.py', 'ustruct.py', 'utime.py', 'utimeq.py', 'uzlib.py', 'websocket.py']
```

I can download the stubs by FTP, but can't understand how to install them manually.

On a seperate note, I'm also happy to create a PR to submit them to the stubs repo, is there a how to on the best way to do this?

BradenM, 2019-10-08:
Hi @jmannau, I apologize for my late reply, I must have missed the notification.

Have you had any luck regarding this issue? I do not have a PyBoard myself, and have been unable to reproduce this issue on any other device so this has been a bit tricky for me. I will continue to look further into this however, as there are many improvements that can be made regarding `micropy-cli`'s automated use of `micropython-stubber`. 

As for adding the stubs, please see the [micropy-stubs](https://github.com/BradenM/micropy-stubs) repo. I know the documentation there is not quite where I would like it to be yet, but feel free to open a PR and I will help out as needed. Be sure to checkout the [example directory](https://github.com/BradenM/micropy-stubs/tree/master/example).

jmannau, 2019-10-08:
Hi Braden,
No, I wasn't able to solve it. I worked around it by mimicking an existing
board stub with the generated stubs.  I will submit a PR to that repo with
these stubs that I generated.
Do you have any pointers on where to look and I can give it another try.
James

BradenM, 2019-10-08:
Well, I do see that your Pyboard has a root of `/flash` which *should* be handled `rshell` if I recall correctly, but perhaps you could fiddle with [this rsync call](https://github.com/BradenM/micropy-cli/blob/5d518d3b535828ffc0b7b8e547c369bff03925a3/micropy/main.py#L102) and [this method](https://github.com/BradenM/micropy-cli/blob/5d518d3b535828ffc0b7b8e547c369bff03925a3/micropy/utils/pybwrapper.py#L45) in `micropy.utils.pybwrapper` and attempt with hard coded values.

I will also merge #58 in a moment (assuming it passes all the tests) which will update the `micropython-stubber` submodule to the latest commit in [my fork](https://github.com/BradenM/micropython-stubber) (also just included your PR on the base repo).

Anyways, I am off to catch some sleep tonight haha, but I will continue this soon. Thanks.

jmannau, 2019-10-08:
So I spent some time looking into this. the `dir_path = self._pyb_path(path)` [link](https://github.com/BradenM/micropy-cli/blob/master/micropy/utils/pybwrapper.py#L192) returns a `/pyboard/stubs` which results in the not found error.  If I manually set the dir_path to `/flash/stubs` the copy gets further. 

Unfortunately, in my case, it then dies with a `binascii.Error: Non-hexadecimal digit found` which is an error from rshell.  Which I'll leave to a different issue.

sfewings, 2019-11-18:
I also have this problem with the Pyboard. Has there been a fix, or someone done a PR for the Pyboard stubs?

BradenM, 2019-11-18:
@sfewings, unfortunately no, sorry. As @jmannau pointed out earlier, the core issue is likely with rshell, and as I do not have a device to test with I have yet to figure this one out. 

I will be looking into alternatives for rshell sometime soon, but for now I would suggest using the closest stub package available (or a combination of them). I would suggest using an esp32 variant that resembles your micropython branch/version. Fortunately several of the libraries are shared between the devices so it shouldn't be lacking too much.

If you do manage to create stubs through the methods jmannau used earlier, feel free to open a PR on [micropy-stubs](https://github.com/BradenM/micropy-stubs).

sfewings, 2019-11-18:
Thanks BradenM, I'm currently using the ESP32 stubs which are indeed very similar. 
Thanks for all your effort with this development. The intellisense is fantastic.

Carglglz, 2020-01-08:
Hi, about this [issue](https://github.com/Carglglz/upydevice/issues/1)
I manage to do a pretty simple 'base serial device class', just needs Pyserial.
```
import time
import ast
import serial

class BASE_SERIAL_DEVICE:
    def __init__(self, serial_port, baudrate):
        self.serial = serial.Serial(serial_port, baudrate)
        self.bytes_sent = 0
        self.buff = b''
    def cmd(self, cmd, decode=True):
        self.bytes_sent = self.serial.write(bytes(cmd+'\r', 'utf-8'))
        time.sleep(0.2)
        self.buff = self.serial.read_all()[self.bytes_sent+1:]
        if decode:
            try:
                return ast.literal_eval(self.buff.replace(b'\r\n',b'').replace(b'>>> ', b'').decode())
            except Exception as e:
                return self.buff.replace(b'\r\n',b'').replace(b'>>> ', b'').decode()
        else:
            try:
                return ast.literal_eval(self.buff.replace(b'\r\n',b'').replace(b'>>> ', b''))
            except Exception as e:
                return self.buff.replace(b'\r\n',b'').replace(b'>>> ', b'')
```

To be used like this:

```
 sdev = BASE_SERIAL_DEVICE('/dev/tty.usbmodem3370377430372', 9600)
 sdev.cmd("print('Hello')")
'Hello'

```
It works with pyboard/esp32/esp8266 (tested so far)
I will implement this in upydevice but right now is still under testing
So if you want to test/implement it yourself feel free to do it ;)
Cheers

mattytrentini, 2020-06-16:
Also see BradenM/micropy-stubs#10. There's a hacky work-around there that allows `micropy stubs create` to run.

Pixel-Therapy, 2020-08-20:
I've got the same bug on a m5 stack 
`Source directory /pyboard/stubs does not exist.`
also all the stubs are there in /stubs folder

edwardjedmonds, 2020-09-09:
I have the same issue on a stock pyboard v1.1 running firmware 1.13

`PS D:\Edward\Documents\GitHub> micropy stubs create COM3

MicroPy  Connecting to Pyboard @ COM3
MicroPy  ✔ Connected!
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Stub module: uasyncio/core
MicroPy  Pyboard  Stub module: umqtt/robust
MicroPy  Pyboard  Stub module: umqtt/simple
MicroPy  Pyboard  Stub module: urllib/urequest
MicroPy  Pyboard  Stub module: _thread
MicroPy  Pyboard  Stub module: ak8963
MicroPy  Pyboard  Stub module: apa102
MicroPy  Pyboard  Stub module: apa106
MicroPy  Pyboard  Stub module: array
MicroPy  Pyboard  Stub module: binascii
MicroPy  Pyboard  Stub module: btree
MicroPy  Pyboard  Stub module: builtins
MicroPy  Pyboard  Stub module: cmath
MicroPy  Pyboard  Stub module: collections
MicroPy  Pyboard  Stub module: crypto
MicroPy  Pyboard  Stub module: curl
MicroPy  Pyboard  Stub module: dht
MicroPy  Pyboard  Stub module: display
MicroPy  Pyboard  Stub module: ds18x20
MicroPy  Pyboard  Stub module: errno
MicroPy  Pyboard  Stub module: esp
MicroPy  Pyboard  Stub module: esp32
MicroPy  Pyboard  Stub module: flashbdev
MicroPy  Pyboard  Stub module: framebuf
MicroPy  Pyboard  Stub module: freesans20
MicroPy  Pyboard  Stub module: functools
MicroPy  Pyboard  Stub module: gc
MicroPy  Pyboard  Stub module: gsm
MicroPy  Pyboard  Stub module: hashlib
MicroPy  Pyboard  Stub module: heapq
MicroPy  Pyboard  Stub module: inisetup
MicroPy  Pyboard  Stub module: io
MicroPy  Pyboard  Stub module: json
MicroPy  Pyboard  Stub module: logging
MicroPy  Pyboard  Stub module: lwip
MicroPy  Pyboard  Stub module: machine
MicroPy  Pyboard  Stub module: math
MicroPy  Pyboard  Stub module: microWebSocket
MicroPy  Pyboard  Stub module: microWebSrv
MicroPy  Pyboard  Stub module: microWebTemplate
MicroPy  Pyboard  Stub module: micropython
MicroPy  Pyboard  Stub module: mpu6500
MicroPy  Pyboard  Stub module: mpu9250
MicroPy  Pyboard  Stub module: neopixel
MicroPy  Pyboard  Stub module: network
MicroPy  Pyboard  Stub module: ntptime
MicroPy  Pyboard  Stub module: onewire
MicroPy  Pyboard  Stub module: os
MicroPy  Pyboard  Stub module: port_diag
MicroPy  Pyboard  Stub module: pycom
MicroPy  Pyboard  Stub module: pye
MicroPy  Pyboard  Stub module: random
MicroPy  Pyboard  Stub module: re
MicroPy  Pyboard  Stub module: requests
MicroPy  Pyboard  Stub module: select
MicroPy  Pyboard  Stub module: socket
MicroPy  Pyboard  Stub module: ssd1306
MicroPy  Pyboard  Stub module: ssh
MicroPy  Pyboard  Stub module: ssl
MicroPy  Pyboard  Stub module: struct
MicroPy  Pyboard  Stub module: sys
MicroPy  Pyboard  Stub module: time
MicroPy  Pyboard  Stub module: tpcalib
MicroPy  Pyboard  Stub module: ubinascii
MicroPy  Pyboard  Stub module: ucollections
MicroPy  Pyboard  Stub module: ucryptolib
MicroPy  Pyboard  Stub module: uctypes
MicroPy  Pyboard  Stub module: uerrno
MicroPy  Pyboard  Stub module: uhashlib
MicroPy  Pyboard  Stub module: uheapq
MicroPy  Pyboard  Stub module: uio
MicroPy  Pyboard  Stub module: ujson
MicroPy  Pyboard  Stub module: uos
MicroPy  Pyboard  Stub module: upip
MicroPy  Pyboard  Stub module: upip_utarfile
MicroPy  Pyboard  Stub module: urandom
MicroPy  Pyboard  Stub module: ure
MicroPy  Pyboard  Stub module: urequests
MicroPy  Pyboard  Stub module: uselect
MicroPy  Pyboard  Stub module: usocket
MicroPy  Pyboard  Stub module: ussl
MicroPy  Pyboard  Stub module: ustruct
MicroPy  Pyboard  Stub module: utime
MicroPy  Pyboard  Stub module: utimeq
MicroPy  Pyboard  Stub module: uwebsocket
MicroPy  Pyboard  Stub module: uzlib
MicroPy  Pyboard  Stub module: websocket
MicroPy  Pyboard  Stub module: websocket_helper
MicroPy  Pyboard  Stub module: writer
MicroPy  Pyboard  Stub module: ymodem
MicroPy  Pyboard  Stub module: zlib
MicroPy  ✔ Done!
MicroPy  Copying stubs...
Source directory /pyboard/stubs does not exist.
Traceback (most recent call last):
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\edwar\AppData\Local\Programs\Python\Python38-32\Scripts\micropy.exe\__main__.py", line 7, in <module>
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\decorators.py", line 73, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\micropy\cli.py", line 280, in create
    return mpy.create_stubs(port, verbose=verbose)
  File "c:\users\edwar\appdata\local\programs\python\python38-32\lib\site-packages\micropy\main.py", line 124, in create_stubs
    stub_path = next(out_dir.iterdir())
StopIteration`

edwardjedmonds, 2020-09-09:
> Also see [BradenM/micropy-stubs#10](https://github.com/BradenM/micropy-stubs/issues/10). There's a hacky work-around there that allows `micropy stubs create` to run.

This worked for me on a stock pyboard v1.1 with 1.13 firmware on windows 10. Thanks.

BradenM, 2020-09-11:
@edwardjedmonds @mattytrentini 

Thank you guys for finding a workaround. As @edwardjedmonds pointed out, seems either `pybwrapper` or `rshell` is failing to determine the correct root path. Will look into this, thanks!

hypermaq, 2020-09-28:
> 
> 
> > Also see [BradenM/micropy-stubs#10](https://github.com/BradenM/micropy-stubs/issues/10). There's a hacky work-around there that allows `micropy stubs create` to run.
> 
> This worked for me on a stock pyboard v1.1 with 1.13 firmware on windows 10. Thanks.

Does that mean you got linting and such things working?

Josverl, 2020-10-23:
If you need the pyboard 1.1 stubs you can find them on https://github.com/Josverl/micropython-stubs

The serial download error with the pyboard is apparent in other software as well, even though upload works well. I've not looked into the root cause yet

hypermaq, 2020-10-27:
> 
> If you need the pyboard 1.1 stubs you can find them on https://github.com/Josverl/micropython-stubs
> 
> The serial download error with the pyboard is apparent in other software as well, even though upload works well. I've not looked into the root cause yet

Thanks. I haven't gotten autocompletion and such working yet, but I'm likely doing something wrong.
I've documented my current steps [here](https://dietervansteenwegen.wordpress.com/2020/09/28/using-pyboard-v1-1-with-visual-studio-code/) but will need to look in to why not everything is working yet. It's rather confusing to me since there seem to be different tools and I don't know what does what exactly.

Josverl, 2020-10-27:
@BradenM 

the code I addedd to pymakr to figure out the root of the board that seems to run on any firmwares that i have tried it on is : 
``` python
import os, errno 
r = "/flash" # assume default 
try: 
    _ = os.stat(r) 
except OSError as e: 
    if e.args[0] == errno.ENOENT: 
        r = os.getcwd() # fallback to current dir , but #current is not always the root of the filesystem
finally: 
    print("'{}'".format(r))
```
that could be added to by trying to find the root 
perhaps something like that will also help to report where the stubs are, or we could reformat the last log line toe report the stub location , so that micropy-cli can read that and be done with it

closed 2022-04-11 by BradenM

Josverl, 2022-04-11:
Interesting ,
How do you handle the issue that `upydev` depends on `netifaces`  requires the VS C++ build tools to be installed ?
does not that make the threshold to install too high ?

```
Building wheels for collected packages: netifaces
  Building wheel for netifaces (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [5 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'netifaces' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for netifaces
  Running setup.py clean for netifaces
Failed to build netifaces
Installing collected packages: netifaces, braceexpand, bleak-winrt, websocket-client, pyusb, Pygments, prompt-toolkit, dill, bleak, argcomplete, bleak-sigspec, upydevice, jupyter-micropython-upydevice, upydev
  Running setup.py install for netifaces ... error
  error: subprocess-exited-with-error

  × Running setup.py install for netifaces did not run successfully.
  │ exit code: 1
  ╰─> [5 lines of output]
      running install
      running build
      running build_ext
      building 'netifaces' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]
````

PR #294 (2022-04-11): PyDevice interfaces, rshell PyDevice backend, new UPYDevice backend.
Adds generic interfaces to enable pydevice communication(s) via pluggable backends.

Along with the migrated rshell backend, adds a new backend that utilizes [Carglglz/upydevice](https://github.com/Carglglz/upydevice).
This newer backend will be used as the default going forward (and always on windows 10 w/ python >=3.10, since rshell currently does not support it).

This will also make implementing features such as #106, #107, #228 considerably easier.

Closes #47
Closes #277