BradenM/micropy-cli #49: Getting jsondecodeerror

ramprao, 2019-09-20:
When I try to install something in VScode , for example, picoweb, I get the following error:

```
$ micropy install picoweb

MicroPy  Loading Project
MicroPy  esp32-micropython-1.11.0 is already installed!

MicroPy  Fetching Requirements
Traceback (most recent call last):
  File "/anaconda3/bin/micropy", line 10, in <module>
    sys.exit(cli())
  File "/anaconda3/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/anaconda3/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/anaconda3/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/anaconda3/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/anaconda3/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/anaconda3/lib/python3.7/site-packages/micropy/cli.py", line 120, in install
    project = Project.resolve('.')
  File "/anaconda3/lib/python3.7/site-packages/micropy/project/project.py", line 388, in resolve
    proj.load(stub_manager=micropy.STUBS, verbose=verbose)
  File "/anaconda3/lib/python3.7/site-packages/micropy/project/project.py", line 195, in load
    self.load_packages()
  File "/anaconda3/lib/python3.7/site-packages/micropy/project/project.py", line 163, in load_packages
    meta = utils.get_package_meta(name, spec=spec)
  File "/anaconda3/lib/python3.7/site-packages/micropy/utils/helpers.py", line 259, in get_package_meta
    data = resp.json()
  File "/anaconda3/lib/python3.7/site-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/anaconda3/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/anaconda3/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/anaconda3/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 7 column 1 (char 6)
```
Get the error with any package I'm trying to install.

BradenM, 2019-09-20:
Hi @ramprao,

Thanks for the report. Looks like the `requests` module is failing to parse the JSON package file from PyPi.

Do you mind providing your `micropy.log` file via a pastebin or similiar?  (Stored in $HOME/.micropy/micropy.log) 

I cannot reproduce this issue, so it may be something wrong with your environment. Please try executing `pip install requests --upgrade` and check if the issue persists. Thanks!

ramprao, 2019-09-20:
Braden, Thanks. Here is the pastebin from the time when I started the micropy command.

(https://pastebin.com/rCVsfm2e)

I did try a reinstall of requests but that did not help. I may try a separate environment.

ramprao, 2019-09-20:
Thanks. You were correct. It works in a new conda environment. Something must have conflicted. 
```
conda create --name "test2"
conda activate test2
conda install pip
conda install requests
pip install micropy 
```
Start VS Code. Change python: Select Interpreter to point to this environment. 
```
micropy stubs add esp32-micropython-1.11.0
micropy init
micropy install picoweb
```
Installed without error.

closed 2019-09-20 by ramprao