BradenM/micropy-cli #68: Error en install package Ultrasonic

fpe, 2019-11-10:
Hi

When I try to install package Ultrasonic, I tog follow errors.

micropy install Ultrasonic

MicroPy  Loading Project
MicroPy  esp32-micropython-1.11.0 is already installed!

MicroPy  ✔ Project Ready!

MicroPy  Installing Packages
MicroPy  Adding Ultrasonic to requirements...
Traceback (most recent call last):
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Flemm\AppData\Local\Programs\Python\Python37\Scripts\micropy.exe\__main__.py", line 9, in <module>
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\micropy\cli.py", line 148, in install
    project.add_package(pkg, dev=dev)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\micropy\project\project.py", line 241, in add_package
    self.load_packages()
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\micropy\project\project.py", line 164, in load_packages
    meta = utils.get_package_meta(name, spec=spec)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\micropy\utils\helpers.py", line 268, in get_package_meta
    data = resp.json()
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\site-packages\requests\models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\json\__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "c:\users\flemm\appdata\local\programs\python\python37\lib\json\decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 7 column 1 (char 6)

BradenM, 2019-11-10:
Hi @fpe,

Can you link me to the package your trying to install? As of this moment, any packages are sourced from [pypi](https://pypi.org) and it looks like there isn't a package available named "Ultrasonic" _(note: packages are case-sensitive_)

However, this does point out the issue that such an error should be properly dealt with rather then exiting with an exception. Will fix soon.

fpe, 2019-11-11:
Hi Braden

First thanks.

I use following Library from  https://github.com/skgsergio/MicropythonLibs.

Med venlig hilsen
Flemming Petersen
Kildebakkegårds Allé 24
2870 Dyssegård
Mobil  : +45 2080 3958
Email : Flemming_Petersen@hotmail.com<mailto:Flemming_Petersen@hotmail.com>

BradenM, 2019-11-11:
Oh I see. I will add support for adding packages from git soon.

Something like this maybe:

`micropy install git+https://github.com/skgsergio/MicropythonLibs/Ultrasonic`

BradenM, 2019-11-11:
Hi @fpe,

While I do still plan on implementing adding packages from Git, it is not likely that this specific repo will work, as it is not setup as a package/module (no `setup.py` or proper structure).

Support for this would simply be too broad, and would have to cover a lot of scenarios. 

I suggest taking a look at [mPython-hcsr04](https://pypi.org/project/mPython-hcsr04/) as an alternative.

closed 2019-11-11 by BradenM

fpe, 2019-11-12:
Hi Braden

Thank you very much for the help, I will look at your suggestion and see if it can be used.
Just a quick question and I don't know if it's pymacr.
I downloaded another packer (BME280) no problems there. When I upload my entire program to esp32, it does not come over. Shouldn't that happen? .

Thanks again for the help and a fantastic program.

Med venlig hilsen / Yours sincerely
Flemming Petersen

BradenM, 2019-11-12:
No problem!

Currently there are more than a few ways to upload external libs with micropython (see [here](https://github.com/BradenM/micropy-cli#project-dependencies)) and as such, I originally decided to only handle the local features (autocomplete/intellisense/etc) for deps and leave uploading the libs to a device up to the user.

 However, I will soon be adding additional options to `micropy install` to allow users to either upload files via a `/lib` folder or freeze modules and rebuild the firmware using [micropy-build](https://github.com/BradenM/micropy-build). 

Thank you for the support :heart:

PR #69 (2019-11-11): fix: Handle Errors when adding Packages
Handle package not found or other unknown errors when adding a package

Closes #68
commit 6a3272d7 (2019-11-11): fix: Handle Errors when adding Packages

Handle package not found or other unknown errors when adding a package

Closes #68