BradenM/micropy-cli #71: Autocompletion and linting not available for libraries in src/lib

MathijsNL, 2019-11-28:
Hi there,

**Problem**

I am having some problems with libraries not getting used for auto completion/
Default sys.path is the root folder and /lib

```
>>> import sys
>>> sys.path
['', '/lib']
```

The resulting .vscode\settings.json and .pylintrc do not have this folder included. 
Currently this results in not getting auto completion and proper linting for user defined libraries.

**Expected behaviour:**
Since the /lib folder on the device is available for imports, it should have working auto completion and linting working properly.

**Possible fix:**
Because user libraries go into the lib folder I propose adding these paths.

What are your thoughts? Let me know!

BradenM, 2019-11-28:
Hi @MathijsNL,

Any dependencies for your project can be installed with: `micropy install <PACKAGE>`

Instead of just adding `src/lib` to the appropriate files, I did it this way to:
* Allow people to structure their project however they want
* Allow people to keep their projects cleaner by not having to publish their requirement source(s) to git
* Generate stubs for any added requirements for intellisense improvements
* Allow people to import packages just as they would on a device (no `sys.path` edits in local workspace)

You can find more info regarding `micropy install` [here](https://github.com/BradenM/micropy-cli#project-dependencies).

I'm completely open to discussion though if you have some possible improvements for this.

Also, when you say "user defined libraries", are you referring to custom libs that are not published to pypi or similiar? If so, I will be adding functionality for installing local packages by path to `micropy install` soon.

MathijsNL, 2019-11-28:
> Also, when you say "user defined libraries", are you referring to custom libs that are not published to pypi or similiar?
Exactly this!

> Instead of just adding `src/lib` to the appropriate files, I did it this way to:
I completely agree with the arguments you give.

My only requirement is to have the src/lib folder added to the .vscode/settings.json. 

My thought was to maybe add it to the main menu.

```
    _template_files = {
        'vscode': CodeTemplate,
        'pylint': PylintTemplate,
        'pymakr': "pymakr.conf",
        'main' : "src/main.py",
        'boot' : "src/boot.py",
        'lib' : "src/lib/__init__.py",
        'gitignore': ".gitignore"
    }

    TEMPLATES = {
        'vscode': (['vscode'], ("VSCode Settings for "
                                "Autocompletion/Intellisense")),
        'pymakr': (['pymakr'], "Pymakr Configuration"),
        'pylint': (['pylint'], "Pylint MicroPython Settings"),
        'gitignore': (['gitignore'], "Git Ignore Template"),
        'bootstrap': (['main', 'boot''], "main.py & boot.py files")
        'libfolder': (['lib'], "add lib folder to support user defined libraries")
    }
```

This way people will get a choice of adding this folder to their project. Although not scalable, it is the a common place for libraries since it is included in the sys.path of micropython devices.

> If so, I will be adding functionality for installing local packages by path to `micropy install` soon.
This might be the best way to go. I will wait till this feature is implemented. Let me know if I can help you in any way.

BradenM, 2019-11-28:
Okay, I definitely see how this could be frustrating with a custom library.

How does this sound?

* Add a config option in `micropy.json` called `private-lib-path`:

micropy.json
```
{
  "name": "NewProject",
  "stubs": {
    "esp32-micropython-1.11.0": "1.2.0"
  },
  "packages": {
      "example-pypi-pkg": "==1.0.0"
  },
  "dev-packages": {
    "micropy-cli": "*"
  },
  "config": {
    "vscode": true,
    "pylint": true,
    "private-lib-path": "src/lib" // Default value
  }
}
```

* Allow pkgs to be installed to private lib via `micropy install <PATH>`

`micropy install ./foobar/some-pkg` or `micropy install ./src/lib/some-pkg`

1.) Add pkg to `private-lib-path` if not already there

2.) Update `micropy.json`

```
{
  "name": "NewProject",
  "stubs": {
    "esp32-micropython-1.11.0": "1.2.0"
  },
  "packages": {
      "example-pypi-pkg": "==1.0.0",
      "example-local-pkg": "exlocalpkg"
  },
  "dev-packages": {
    "micropy-cli": "*"
  },
  "config": {
    "vscode": true,
    "pylint": true,
    "private-lib-path": "src/lib"
  }
}

```

where "exlocalpkg" would be a path relative to `private-lib-path`. So in this cause the full path to `example-local-pkg` would be: `my-micropy-project/src/lib/exlocalpkg`

Would this cover your use cases?

MathijsNL, 2019-11-29:
I think that will definitely do the job!

closed 2019-12-09 by BradenM

BradenM, 2019-12-09:
Sorry for the delay! Had a lot of other items I needed to complete for this. 

Not entirely sure the stability of this, so I went ahead and made a pre-release for it. You can try it out by executing: `pip install micropy-cli --upgrade --pre`

Let me know if you run into any issues with it! Thanks.

MathijsNL, 2019-12-09:
No problem, I appreciate you taking the time to implement this feature.

I installed a new project and tried to install custom packages. Works like a charm! Things I checked:
- [x] .vscode/settings.json looks correct, intellisense is working
- [x] requirements.txt looks correct, but when running **micropy install** afterwards it gives some errors
- [x] .pylintrc looks correct, pylint is working

I will try to look into the problem with the requirements.txt tonight or tomorrow hopefully.

BradenM, 2019-12-09:
Thanks for the report!

Feel free to open any new issues when you have the time. Thanks again.

PR #75 (2019-12-09): feat: Add Project Dependencies from Path
Closes #71 

Allows user to add project dependencies via a local path.

This umbrella PR actually includes numerous improvements to the overall project structure:

- [x] Project Config overhauled to use `dpath` library for merging/updating/handling configuration state. 
- [x] Removed per-module "subconfigs/substates" by always referring to the parent config as the one "true state"
- [x] Improved Several tests
- [x] Implements new Packages API into `PackagesModule`
- [x] Some Code Cleanup

[Autocompletion and linting not available for libraries in src/li [... cut]