BradenM/micropy-cli #78: AttributeError: 'TemplatesModule' object has no attribute 'provider'

CeKl, 2019-12-02:
Hi,
in the beginning, thank you for the work you're putting in here.
I work under Win10 in VS code with the esp32 and get the error message "AttributeError: 'TemplatesModule' object has no attribute 'provider'" when I try to create a new project. What can I do about it? I didn't find anything in the code spontaneously.

BradenM, 2019-12-02:
Hi @CeKl,

That's strange, thanks for opening an issue about this.

Do you mind attaching your `micropy.log` file? It should be located under: `$HOME/.micropy/micropy.log` 

Also, can you provide the following:
* Python Version 
* MicropyCli Version (`micropy --version`)

Thanks!

CeKl, 2019-12-02:
Hi @BradenM ,
here's the log file:

[micropy.log](https://github.com/BradenM/micropy-cli/files/3911482/micropy.log)

I use the following:
Python 3.6.8
micropython 1.11 
MicropyCli Version 3.1.0

Already thank you for the quick answer

BradenM, 2019-12-02:
Okay, I'm pretty sure I see the problem. 

Looks like your creating a new project without any templates, is this accurate? If so I believe this is causing the issue.

As a workaround, try creating a project with at least one template.

If this works, I will fix this problem soon.

CeKl, 2019-12-02:
Hi @BradenM ,

when i specify vscode as template i get the error message " TypeError: __init__() got an unexpected keyword argument 'text' " but with pymakr as template it works as expected.

Thank you for your help

BradenM, 2019-12-02:
Okay then my suspucion was correct. 

The `TypeError` you mentioned has been fixed as of da6b56ba6f097c9b7e9d0315062eb93865a5e05a

I will be cherry-picking these and some other recent fixes into a new release soon. Thanks for your help.

CeKl, 2019-12-02:
Thanks for the quick help

commit 7c86b86e (2019-12-03): fix(project): Exception Raised if no Templates are used in Project

Micropy failed to create a new project (raised an AttributeError) if the user did not select any tem
plates to render.

Fixes #78
commit 43a09eb7 (2019-12-03): fix(project): Exception Raised if no Templates are used in Project

Micropy failed to create a new project (raised an AttributeError) if the user did not select any tem
plates to render.

Fixes #78
closed 2019-12-03 by BradenM