BradenM/micropy-cli #79: Organize Tests by Module

BradenM, 2019-12-04:


Josverl, 2020-01-07:
happy to pitch in. im familiar with TDD , but not on Python.
any sample to get me started ?

BradenM, 2020-01-08:
Sure!

So the main issue with the tests right now is that they are simply poorly structured and tend to be very susceptible to [hidden test dependencies](https://testmon.org/hidden-test-dependencies.html).

I'd like to eliminate any [flakyness](https://docs.pytest.org/en/latest/flaky.html), but thats a tough goal currently haha.

Would also like to split the test up into individual modules (see [Good Integration Practices](https://docs.pytest.org/en/latest/goodpractices.html#)).

Overall I'd say take a look at the tests for the [packages module](https://github.com/BradenM/micropy-cli/blob/master/tests/test_packages.py) to get an idea.

Any efforts would be appreciated! Am trying to find time to wrap up #97 and improve overall documentation. 

Thanks.