BradenM/micropy-cli #96: Improve pylint path syntax and add ./src/lib

Josverl, 2019-12-29:
Hi Braden,

I would like to suggest :

1. a change in syntax to the generated `.pylintrc` file to make it cleared to read and modify.
2. the addition of src/lib path as this is commonly used 

**current syntax (windows paths )** 
``` ini
[MASTER]
# Loaded Stubs:  esp32-micropython-1.11.0 
init-hook='import sys;sys.path.insert(1, ".micropy\\BradenM-micropy-stubs-7a7d171\\frozen");sys.path.insert(1, ".micropy\\BradenM-micropy-stubs-452a204\\frozen");sys.path.insert(1, ".micropy\\BradenM-micropy-stubs-7a7d171\\stubs");sys.path.insert(1,"src\\lib");sys.path.insert(1,"src\\font")'

...
```

**new syntax ** 
``` ini
[MASTER]
# Loaded Stubs:  esp32-micropython-1.11.0 
init-hook='import sys;sys.path[1:1] = ["src/lib", "src/fonts", ".micropy/BradenM-micropy-stubs-7a7d171/frozen", ".micropy/BradenM-micropy-stubs-452a204/frozen", ".micropy/BradenM-micropy-stubs-7a7d171/stubs",".micropy/thermo"]'
# Note:  when adding paths; also add these paths to: VSCode Preferences > Workspace > python.autoComplete.extraPaths

...
```
The new syntax has the paths in the right order , is shorter, is simpler to extend with additional paths.

For me this helps in organizing my modules. I think this should be useful to others as well.

/Jos

BradenM, 2019-12-29:
Sure, I like this idea. I could also add an additional config option `extraPaths` that would take an array of paths to add to `.pylintrc` and `python.autoComplete.extraPaths`.

Thoughts?

Josverl, 2020-01-02:
yes, such an extrapaths would be a nice to have.
i think documentation is also relvant if adding that , as it haunted me for quite some time why my `src/lib` folder was not included while the `scr` folder was.
it seems that the Python Language Service has some magic to add a scr folder to the extrapaths even if yo do not specify it.

BradenM, 2020-01-03:
Agreed. Documentation definitely needs some work. Will try and update it soon. Have not added the configuration option yet, but would love to hear your thoughts on #97. Thanks.

PR #174 (2020-10-29): template update for pylint rc
I finally managed to figure out how to download / configure / run and debug this in my own machine 
( ill try to add some hard learned how-to documentation ) 

after that it turned out to be a simple change that I hope you can merge to close #96 even if that is just an in-between step to the other changes you started in #97 

changes are only to the jijnja template file 
- no code added, just a keyword change to align to template
- no tests added, but a few tweaks to recognize the new template
closed 2020-10-29 by BradenM