BradenM/micropy-cli #98: ms-python.python not recognized

WoMeijerRoetz, 2020-03-17:
When running MicroPy init, I get the following error:
```
MicroPy  VSCode Extension ms-python.python failed to satisfy requirements!
MicroPy  Min Required Version: 2019.9.34474
MicroPy  Current Version: 0.0.0
MicroPy  VSCode Integration will fail! See BradenM/micropy-cli#50 for details.
```
despite having 2020.2.64397 installed.

After running the init I get prompted to select a Python interpreter, which is possibly why it's not found?

* OS: Mac OSx Catalina 10.15.3 
* Micropy Version: 3.3.0
* Python Version: 3.7.3
* VSCode Version: 1.43.0

BradenM, 2020-03-17:
Strange. This check should actually be disabled IIRC ...

Try starting your project with:
```sh
$ micropy --skip-checks init
```

And see if that works.

BradenM, 2020-03-31:
closing as stale

closed 2020-03-31 by BradenM

andrewbunday, 2020-04-20:
This looks to be a problem with VSCode not automatically installing the `code` command line on MacOS. So the check is running but errors:
```
$ code --list-extensions --show-versions
bash: code: command not found
```
Running `Shift-Cmd-P` and `Install 'code' command in PATH` resolves the error. It may be worth checking the return code on the subprocess returns 0.