BradenM/micropy-cli #103: Can't resolve import?

rmaes4, 2020-04-18:
Thank you for this project. I am a novice when it comes to micropython and I found your tool to help me get going quickly. I am trying to use the micropython bluetooth library. VS Code is giving me an error however saying that it cannot be found. I tried running `micropy install bluetooth` as well but it said it could not be found. What exactly do I need to do to use the bluetooth library. Please see the screenshot below. Thanks again.
[image]

BradenM, 2020-05-07:
Hi @rmaes4 , sorry for the delay!

This is a known issue, please see BradenM/micropy-stubs#9 for more info. Thanks!

askpatrickw, 2020-05-15:
I opened an issue in the appropriate repo to track this: https://github.com/Josverl/micropython-stubber/issues/13

BradenM, 2021-05-17:
Closed in https://github.com/Josverl/micropython-stubber/issues/13

closed 2021-05-17 by BradenM