BradenM/micropy-cli #108: micropy overwrites existing .vscode/settings.json

askpatrickw, 2020-05-18:
**Describe the bug**
`micropy init` will overwrite `.vscode/settings.json` in a workspace with existing settings

**Expected Behavior**
Micropy should add itself to the workspace settings, only overwriting specific settings entries.

**Current Behavior**
micropy over writes the existing `.vscode/settings.json` loosing user preferences.

**Steps to Reproduce**

1. create a new folder.
2. launch code from that folder `code .`
3. From code make a change that creates a workspace settings file. For example use Peacock to change the IDE color. At this point I have the following settings.json:

```json
{
    "workbench.colorCustomizations": {
        "activityBar.activeBackground": "#65c89b",
        "activityBar.activeBorder": "#945bc4",
        "activityBar.background": "#65c89b",
        "activityBar.foreground": "#15202b",
        "activityBar.inactiveForeground": "#15202b99",
        "activityBarBadge.background": "#945bc4",
        "activityBarBadge.foreground": "#e7e7e7",
        "statusBar.background": "#42b883",
        "statusBar.border": "#42b883",
        "statusBar.foreground": "#15202b",
        "statusBarItem.hoverBackground": "#359268",
        "titleBar.activeBackground": "#42b883",
        "titleBar.activeForeground": "#15202b",
        "titleBar.border": "#42b883",
        "titleBar.inactiveBackground": "#42b88399",
        "titleBar.inactiveForeground": "#15202b99"
    },
    "peacock.color": "#42b883"
}
```

4. Back to the CLI, run `micropy init .` and include the VS Code settings in your choices. I now have the follwing settings.json file:

``` json
{
    "python.linting.enabled": true,
    "python.jediEnabled": false,

    // Loaded Stubs:  esp32-micropython-1.12.0 
    "python.autoComplete.extraPaths": [".micropy/BradenM-micropy-stubs-9569403/frozen", ".micropy/BradenM-micropy-stubs-27ee48a/frozen", ".micropy/BradenM-micropy-stubs-9569403/stubs", ".micropy/bug-repo"],
    "python.autoComplete.typeshedPaths":  [".micropy/BradenM-micropy-stubs-9569403/frozen", ".micropy/BradenM-micropy-stubs-27ee48a/frozen", ".micropy/BradenM-micropy-stubs-9569403/stubs", ".micropy/bug-repo"],
    "python.analysis.typeshedPaths":  [".micropy/BradenM-micropy-stubs-9569403/frozen", ".micropy/BradenM-micropy-stubs-27ee48a/frozen", ".micropy/BradenM-micropy-stubs-9569403/stubs", ".micropy/bug-repo"],

    "python.linting.pylintEnabled": true
}
```

**Possible Solution**
NOTE: If you have a main.py already in the workspace it will not be overwritten for that init option.

**Logs**
N\A

**Context (Environment)**

* OS: MacOS
* Micropy Version:  micropy, version 3.3.0
* Python Version: 3.8.2
* VSCode Version: 1.45.1 - 5763d909d5f12fe19f215cbfdd29a91c0fa9208a - x64

BradenM, 2020-05-22:
Am I correct in saying this only happens when adding micropy to a pre-existing workspace?

If so, then the solution would likely be to add a check to `TemplateProvider` somewhere around [here](https://github.com/BradenM/micropy-cli/blob/b6f8efc2108006bf347b8360993e270713eb0c8d/micropy/project/template.py#L307) to see if the settings file already exists, and if so update it as it does usually (post-creation).

askpatrickw, 2020-05-22:
I'll have a look at this... but first figuring out how to build...
I'll open a separate Issue on how to build

PR #112 (2020-05-24): Update existing settings
Fixes #108 
Pretty straight-forward change, now that I can build ;-)
Check if the file does not exist, then create it (the mainline case and uses the previous code)
If the file does exist, then update it instead.
closed 2020-05-24 by BradenM