BradenM/micropy-cli #110: How To Build

askpatrickw, 2020-05-22:
I want to get the steps and dependencies nailed down to run and install a local build. Happy to add to the docs once this is figured out.

I think I'm missing a dependency or a setting or something as I'm seeing errors and micropy does not run. Let me know if you see something obvious.

At the end of all the steps below, micropy is unrecognized (but it is in pip list) and running `python -m micropy help` from the root of the project results in the following tracback:

[command-not-found-micropy.log](https://github.com/BradenM/micropy-cli/files/4670126/command-not-found-micropy.log)

```
micropy-cli git:(master) ✗ pwd                   
/Users/askpatrickw/code/github/micropy-cli
➜  micropy-cli git:(master) ✗ python -m micropy help
Traceback (most recent call last):
  File "/Users/askpatrickw/.pyenv/versions/3.8.1/lib/python3.8/runpy.py", line 184, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/Users/askpatrickw/.pyenv/versions/3.8.1/lib/python3.8/runpy.py", line 143, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/Users/askpatrickw/.pyenv/versions/3.8.1/lib/python3.8/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/__init__.py", line 18, in <module>
    from micropy import data, lib, project, stubs, utils
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/project/__init__.py", line 5, in <module>
    from . import modules
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/project/modules/__init__.py", line 6, in <module>
    from .modules import HookProxy, ProjectModule
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/project/modules/modules.py", line 9, in <module>
    from micropy import utils
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/utils/__init__.py", line 11, in <module>
    from . import decorators, helpers
  File "/Users/askpatrickw/code/github/micropy-cli/micropy/utils/helpers.py", line 31, in <module>
    from micropy.lib.stubber.runOnPc import make_stub_files as stubgen
ModuleNotFoundError: No module named 'micropy.lib.stubber.runOnPc'
```

Based on the Github Actions Logs, here's how I did the build:

- I'm running this from a virtualenv (I am using Pyenv on MacOS with Py3.8.1) NOTE: I clean installed 3.8.1 and created a new virtualenv to ensure I had no conflicts 
- You must have a version of Python in .python_versions in your virtualenv (I added 3.8.1 to .python_versions file and the TOX file)
- Must have [Poetry installed](https://python-poetry.org/docs/)
- Run `poetry install -v` (which will also build micropy-cli)
- Run `poetry run tox` to run tests (it will fail for any python versions you don't have installed)

I ran TOX limited it to py38 and I see the following errors which I also see in the [PR runs](https://github.com/BradenM/micropy-cli/runs/654701364?check_suite_focus=true#step:6:18)

```
py38 run-test: commands[1] | poetry install -E create_stubs -v
WARNING: test command found but not installed in testenv
  cmd: /Users/askpatrickw/.poetry/bin/poetry
  env: /Users/askpatrickw/code/github/micropy-cli/.tox/py38
Maybe you forgot to specify a dependency? See also the whitelist_externals envconfig setting.
```
AND [also in PR Runs](https://github.com/BradenM/micropy-cli/runs/654701364?check_suite_focus=true#step:6:138)
```
py38 run-test: commands[2] | poetry run pytest --cov --cov-config=setup.cfg --junit-xml=test_log.xml --cov-report=xml:cov.xml -vv -ra -n0
WARNING: test command found but not installed in testenv
  cmd: /Users/askpatrickw/.poetry/bin/poetry
  env: /Users/askpatrickw/code/github/micropy-cli/.tox/py38
Maybe you forgot to specify a dependency? See also the whitelist_externals envconfig setting.

DEPRECATION WARNING: this will be an error in tox 4 and above!
ImportError while loading conftest '/Users/askpatrickw/code/github/micropy-cli/tests/conftest.py'.
tests/conftest.py:14: in <module>
    import micropy
micropy/__init__.py:18: in <module>
    from micropy import data, lib, project, stubs, utils
micropy/project/__init__.py:5: in <module>
    from . import modules
micropy/project/modules/__init__.py:6: in <module>
    from .modules import HookProxy, ProjectModule
micropy/project/modules/modules.py:9: in <module>
    from micropy import utils
micropy/utils/__init__.py:11: in <module>
    from . import decorators, helpers
micropy/utils/helpers.py:31: in <module>
    from micropy.lib.stubber.runOnPc import make_stub_files as stubgen
E   ModuleNotFoundError: No module named 'micropy.lib.stubber.runOnPc'
ERROR: InvocationError for command /Users/askpatrickw/.poetry/bin/poetry run pytest --cov --cov-config=setup.cfg --junit-xml=test_log.xml --cov-report=xml:cov.xml -vv -ra -n0 (exited with code 4)
_________________ summary _________________ 
ERROR:   py38: commands failed
```

This was pip list before running poetry
```
$ pip list         
Package    Version
---------- -------
pip        20.1.1
setuptools 41.2.0

```

Full Logs
[poetry-run-tox.log](https://github.com/BradenM/micropy-cli/files/4670086/poetry-run-tox.log)

I'm going to keep digging, but seems like something obvious is missing that I'm not seeing. Thanks!

askpatrickw, 2020-05-22:
OK... the TOX warnings are probably not an issue now. I'll update the PR when I learn more.

I am still unable to run micropy though.

After all that, I ran poetry build.
```
poetry build       
Building micropy-cli (3.3.0)
 - Building sdist
 - Built micropy-cli-3.3.0.tar.gz

 - Building wheel
 - Built micropy_cli-3.3.0-py3-none-any.whl
```

I installed form the wheel `pip install dist/micropy_cli-3.3.0-py3-none-any.whl`

And I still get a traceback

```

askpatrickw, 2020-05-22:
Didn't mean to close that.

I also tried runing make, make install, make dist.. intalling the whl. Same lib error.

I'm going to stop for now. I'm stuck. :-(

BradenM, 2020-05-23:
Hey @askpatrickw, sorry to hear you're having some trouble w/ it.

I will add instructions on setting up a local dev env for micropy-cli soon, but the quick answer to your issue is that `micropy-cli` depends on [my fork](https://github.com/BradenM/micropython-stubber/tree/pkg/slim) of Josverl's micropython-stubber and is installed as a [git submodule](https://github.com/BradenM/micropy-cli/blob/master/.gitmodules).

So, after cloning micropy-cli, run:
`git submodule update --init --recursive` 

This will pull down micropython-stubber into the `lib/` subpackage for you and everything should work fine from there. Feel free to let me know if you run into any other troubles! Thanks.

closed 2020-05-23 by BradenM

askpatrickw, 2020-05-23:
Bingo! That was the magic sauce. Thanks Braden!