BradenM/micropy-cli #135: src/lib is ignored from version control

askpatrickw, 2020-06-16:
**Describe the bug**
src/lib is ignored by version control

**Expected Behavior**
files added to src/lib should be checked in.

**Current Behavior**
They are ignored.

**Work Around && Fix **

add `!src/lib` to bottom of .gitignore

BradenM, 2020-06-18:
Actually, with the mindset of "src/lib" == 3rd party dependencies, this was on purpose. 

Ideally, I'd like to have micropy rebuild this per person from the micropy.json file ( as it does currently with the .micropy/ stubs and such). 

Do you have different intentions in mind?

askpatrickw, 2020-06-18:
Unfortunately, many (most?) MicroPython libraries can only be manually copied into a project.
People just don't go through the steps to publish them as packages.

That was my thinking.

BradenM, 2020-06-19:
Ah, I gotcha. 

What do you think about being able to install packages directly from git instead?

Something like:
   `micropy install  git+https://github.com/user/project.git#egg=project`

askpatrickw, 2020-06-19:
I'm aware of all these ways to manage code.

If you don't want to take the PR, that's cool, its just a suggestion.

PR #137 (2020-08-03): Track changes in src/lib
fixes #135
commit e8707cf0 (2020-08-03): Track changes in src/lib

fixes #135
closed 2020-08-03 by BradenM