BradenM/micropy-cli #152: Pylance Support

askpatrickw, 2020-07-25:
This is probably more a research project, but I wanted to put it out there to get input.

https://marketplace.visualstudio.com/items?itemName=ms-python.vscode-pylance

> Pylance is an extension that works alongside Python in Visual Studio Code to provide performant language support. Under the hood, Pylance is powered by Pyright, Microsoft's static type checking tool. Using Pyright, Pylance has the ability to supercharge your Python IntelliSense experience with rich type information, helping you write better code faster.

Micropy-cli uses :

- python.autoComplete.extraPaths
- python.autoComplete.typeshedPaths
- python.analysis.typeshedPaths

I wonder if Micropy-cli should also set similar sounding Pylance Settings

- python.analysis.stubPath
- python.analysis.autoSearchPaths
- python.analysis.extraPaths

Like I said, I'm not 100%. What do you think?

BradenM, 2020-07-25:
Very interesting. Have you personally tried this? If so, how did it compare to the current ms-python + micropy stubs combo?

I've been very late putting out a release with the recent changes, so I will try and get that done today and then take a deeper look into this.

On a side note, @hlovatt recently contacted me with a new project he's working on that generates stubs with docstrings parsed straight out of micropython's documentation. 

See:

[hlovatt/PyBoardTypeshedGenerator](https://github.com/hlovatt/PyBoardTypeshedGenerator) 
And it's output:
[hlovatt/PyBoardTypeshed](https://github.com/hlovatt/PyBoardTypeshed)

askpatrickw, 2020-07-26:
I've not played around with it yet in micropython

I believe Circuit Python uses a similar method to generate stubs
https://github.com/adafruit/circuitpython/issues?q=is%3Aissue+is%3Aopen+stubs

ShenTengTu, 2020-09-15:
`python.analysis.extraPaths` in Pylance equivalent to current `python.autoComplete.extraPaths`.

this setting works well for me.

krenzlin, 2020-11-02:
While Pylance (v2020.10.3) now finds the stubs, it misses `range` and emits `"range" is not definedPylance (reportUndefinedVariable)`

Josverl, 2020-11-07:
i've  been looking into this from micropython-stubber to understand what needs to be done to get Lancelot happy.
i have not cracked the code on this yet.
i have a suspicion that pylance gets thrown off by the CPython-stubs, but no hard evidence yet 
it would be good to have a minimal repo of where it goes wrong , so that we can whittle it down and hopefully sort it.

@ShenTengTu , as yo seem to have a working config , could you share your setup please ?

CallumJHays, 2020-12-01:
Hey, Just gave this a quick try. removing all options but "python.analysis.extraPaths" seems to be all thats needed for setting up with pylance. Setting "python.analysis.typeShedPaths" as in the generated config appears to cause pylance's inferencing to play up.
 
Working config:
```json
{
  "python.linting.enabled": true,
  "python.analysis.extraPaths": [
    ".micropy/BradenM-micropy-stubs-3cc928b/frozen",
    ".micropy/BradenM-micropy-stubs-e1b8ce6/frozen",
    ".micropy/BradenM-micropy-stubs-3cc928b/stubs",
    ".micropy/bdsim.micropython"
  ],
  "python.linting.pylintEnabled": true
}
```

Josverl, 2020-12-01:
Thanks @CallumJHays , I can confirmthat pylint works better when dropping the additional paths 

it may be needed to add an explicit   `"python.languageServer": "Pylance",`  depending on the user / ws preferences 

```
{
    "python.languageServer": "Pylance",
    "python.autoComplete.extraPaths": [
        ".micropy/esp32_p1_meter",
        "src/lib",
        "all-stubs/cpython_patch",
        "all-stubs/mpy_1_13-nightly_frozen/esp32/GENERIC", 
        "all-stubs/esp32_1_13_0-103",
    ],
    "python.linting.enabled": true,
    "python.linting.pylintEnabled": true,
}
```
Ill update the upstream documentation tracked by : https://github.com/Josverl/micropython-stubber/issues/35

BradenM, 2020-12-01:
Awesome! 

Is anyone here interested in pulling together a PR to add a template option for pylance in micropy-cli?

If not, I will try to get to it soon.

PR #379 (2023-01-30): VSCode Pylance Support
- Add Pylance support Fix #152
- feat(project): add pylance settings to vscode template.
- feat(project): assume pylance until proper refactorings can be done.
commit 70a00418 (2023-01-30): Add Pylance support
Fix #152
closed 2023-01-30 by BradenM