BradenM/micropy-cli #167: stubs create panics ESP32 running Micropython 1.13

cabletie, 2020-09-26:
### Question:
The first question is really is there a stub for MP 1.13?
The search doesn't return one...
So I tried creating one but it causes the ESP32 to panic (see logs below).
It seems to create just the /pyboard/stubs/esp32_1_13_0/uasyncio/core.py but that's all and it doesn't seem to download anything.

Thanks for any help.

---
### Logs:
$ micropy stubs create /dev/tty.SLAB_USBtoUART --verbose

MicroPy  Connecting to Pyboard @ /dev/tty.SLAB_USBtoUART
Connecting to /dev/tty.SLAB_USBtoUART (buffer-size 512)...
Trying to connect to REPL  connected
Testing if sys.stdin.buffer exists ... Y
Retrieving root directories ... /boot.py/ /lib/ /main.py/ /mynetwork.py/ /novalight.py/ /np.py/ /secrets.py/ /stubs/
Setting time ... Sep 26, 2020 18:18:02
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
MicroPy  ✔ Connected!
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Stub module: uasyncio/core
MicroPy  Pyboard  Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
MicroPy  Pyboard  Core 1 register dump:
MicroPy  Pyboard  PC      : 0x400e8efe  PS      : 0x00060330  A0      : 0x800df481  A1      : 0x3ffcdf50
MicroPy  Pyboard  A2      : 0x00000002  A3      : 0x000000b3  A4      : 0x3ffce030  A5      : 0x00000000
MicroPy  Pyboard  A6      : 0x000000b3  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffcdf50
MicroPy  Pyboard  A10     : 0x00000000  A11     : 0x3ffba28c  A12     : 0x00000001  A13     : 0x00000000
MicroPy  Pyboard  A14     : 0x3f40bbc0  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
MicroPy  Pyboard  EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
MicroPy  Pyboard  
MicroPy  Pyboard  ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
MicroPy  Pyboard  
MicroPy  Pyboard  Backtrace: 0x400e8efe:0x3ffcdf50 0x400df47e:0x3ffcdf90 0x400df4f0:0x3ffcdfb0 0x400eac51:0x3ffce030 0x400e380e:0x3ffce060 0x400df75d:0x3ffce090 0x400ecf45:0x3ffce0b0 0x400e38c8:0x3ffce150 0x400df75d:0x3ffce180 0x400df885:0x3ffce1a0 0x400ecfcd:0x3ffce1c0 0x400e38c8:0x3ffce260 0x400df75d:0x3ffce290 0x400df885:0x3ffce2b0 0x400ecfcd:0x3ffce2d0 0x400e38c8:0x3ffce370 0x400df75d:0x3ffce3a0 0x400df885:0x3ffce3c0 0x400ecfcd:0x3ffce3e0 0x400e38c8:0x3ffce480 0x400df75d:0x3ffce4b0 0x400df885:0x3ffce4d0 0x400ecfcd:0x3ffce4f0 0x400e38c8:0x3ffce590 0x400df75d:0x3ffce5c0 0x400df885:0x3ffce5e0 0x400ecfcd:0x3ffce600 0x400e38c8:0x3ffce6a0 0x400df75d:0x3ffce720 0x400ecf45:0x3ffce740 0x400e38c8:0x3ffce7e0 0x400df75d:0x3ffce850 0x400df786:0x3ffce870 0x4010482f:0x3ffce890 0x40104921:0x3ffce920 0x400f5cbd:0x3ffce950 0x4009b022:0x3ffce980
MicroPy  Pyboard  
MicroPy  Pyboard  Rebooting...
MicroPy  Pyboard  ets Jun  8 2016 00:22:57
MicroPy  Pyboard  
MicroPy  Pyboard  rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
MicroPy  Pyboard  configsip: 0, SPIWP:0xee
MicroPy  Pyboard  clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
MicroPy  Pyboard  mode:DIO, clock div:2
MicroPy  Pyboard  load:0x3fff0018,len:4
MicroPy  Pyboard  load:0x3fff001c,len:5008
MicroPy  Pyboard  ho 0 tail 12 room 4
MicroPy  Pyboard  load:0x40078000,len:10600
MicroPy  Pyboard  ho 0 tail 12 room 4
MicroPy  Pyboard  load:0x40080400,len:5684
MicroPy  Pyboard  entry 0x400806bc
MicroPy  Pyboard  I (548) cpu_start: Pro cpu up.
MicroPy  Pyboard  I (548) cpu_start: Application information:
MicroPy  Pyboard  I (548) cpu_start: Compile time:     Sep  2 2020 03:00:08
MicroPy  Pyboard  I (551) cpu_start: ELF file SHA256:  0000000000000000...
MicroPy  Pyboard  I (557) cpu_start: ESP-IDF:          v3.3.2
MicroPy  Pyboard  I (562) cpu_start: Starting app cpu, entry point is 0x40082f30
MicroPy  Pyboard  I (552) cpu_start: App cpu up.
MicroPy  Pyboard  I (573) heap_init: Initializing. RAM available for dynamic allocation:
MicroPy  Pyboard  I (580) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
MicroPy  Pyboard  I (586) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
MicroPy  Pyboard  I (592) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
MicroPy  Pyboard  I (598) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
MicroPy  Pyboard  I (604) heap_init: At 3FFCA9E8 len 00015618 (85 KiB): DRAM
MicroPy  Pyboard  I (610) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
MicroPy  Pyboard  I (616) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
MicroPy  Pyboard  I (623) heap_init: At 4009DE28 len 000021D8 (8 KiB): IRAM
MicroPy  Pyboard  I (629) cpu_start: Pro cpu start user code
MicroPy  Pyboard  I (312) cpu_start: Starting scheduler on PRO CPU.
MicroPy  Pyboard  I (0) cpu_start: Starting scheduler on APP CPU.
MicroPy  Pyboard  MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32
MicroPy  Pyboard  Type "help()" for more information.
MicroPy  Failed to execute script: timeout waiting for first EOF reception

persello, 2020-10-21:
Same for me, even when compiling Micropython with IDF v4.0.

```
MicroPy  Connecting to Pyboard @ /dev/ttyUSB0
MicroPy  ✔ Connected!
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Stub module: uasyncio/core
MicroPy  Pyboard  Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
MicroPy  Pyboard  Core 1 register dump:
MicroPy  Pyboard  PC      : 0x400e9e46  PS      : 0x00060d30  A0      : 0x800dfded  A1      : 0x3ffce230
MicroPy  Pyboard  A2      : 0x3f40b878  A3      : 0x00000002  A4      : 0x3ffce310  A5      : 0x00000000
MicroPy  Pyboard  A6      : 0x000000b3  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffce230
MicroPy  Pyboard  A10     : 0x00000000  A11     : 0x3ffba184  A12     : 0x00000000  A13     : 0x00000000
MicroPy  Pyboard  A14     : 0x00000001  A15     : 0x3f40b878  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
MicroPy  Pyboard  EXCVADDR: 0x00000004  LBEG    : 0x40096908  LEND    : 0x40096913  LCOUNT  : 0x00000000
MicroPy  Pyboard  
MicroPy  Pyboard  ELF file SHA256: 0000000000000000
MicroPy  Pyboard  
MicroPy  Pyboard  Backtrace: 0x400e9e43:0x3ffce230 0x400dfdea:0x3ffce270 0x400dfe58:0x3ffce290 0x400ebd59:0x3ffce310 0x400e43a2:0x3ffce340 0x400e00d5:0x3ffce370 0x400ee50a:0x3ffce390 0x400e4464:0x3ffce430 0x400e00d5:0x3ffce460 0x400e0216:0x3ffce480 0x400ee583:0x3ffce4a0 0x400e4464:0x3ffce540 0x400e00d5:0x3ffce570 0x400e0216:0x3ffce590 0x400ee583:0x3ffce5b0 0x400e4464:0x3ffce650 0x400e00d5:0x3ffce680 0x400e0216:0x3ffce6a0 0x400ee583:0x3ffce6c0 0x400e4464:0x3ffce760 0x400e00d5:0x3ffce790 0x400e0216:0x3ffce7b0 0x400ee583:0x3ffce7d0 0x400e4464:0x3ffce870 0x400e00d5:0x3ffce8a0 0x400e0216:0x3ffce8c0 0x400ee583:0x3ffce8e0 0x400e4464:0x3ffce980 0x400e00d5:0x3ffce9f0 0x400ee50a:0x3ffcea10 0x400e4464:0x3ffceab0 0x400e00d5:0x3ffceb10 0x400e00fe:0x3ffceb30 0x40104470:0x3ffceb50 0x40104565:0x3ffcebe0 0x400f7d3d:0x3ffcec10 0x40092b15:0x3ffcec40
MicroPy  Pyboard  
MicroPy  Pyboard  Rebooting...
MicroPy  Pyboard  I (561) psram: This chip is ESP32-D0WD
MicroPy  Pyboard  I (561) spiram: Found 64MBit SPI RAM device
MicroPy  Pyboard  I (561) spiram: SPI RAM mode: flash 40m sram 40m
MicroPy  Pyboard  I (564) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
MicroPy  Pyboard  I (571) cpu_start: Pro cpu up.
MicroPy  Pyboard  I (575) cpu_start: Application information:
MicroPy  Pyboard  I (580) cpu_start: Compile time:     Oct 21 2020 16:32:00
MicroPy  Pyboard  I (586) cpu_start: ELF file SHA256:  0000000000000000...
MicroPy  Pyboard  I (592) cpu_start: ESP-IDF:          v4.0.1
MicroPy  Pyboard  I (597) cpu_start: Starting app cpu, entry point is 0x40082db4
MicroPy  Pyboard  I (597) cpu_start: App cpu up.
MicroPy  Pyboard  I (1475) spiram: SPI SRAM memory test OK
MicroPy  Pyboard  I (1475) heap_init: Initializing. RAM available for dynamic allocation:
MicroPy  Pyboard  I (1476) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
MicroPy  Pyboard  I (1482) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
MicroPy  Pyboard  I (1488) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
MicroPy  Pyboard  I (1494) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
MicroPy  Pyboard  I (1500) heap_init: At 3FFCACA8 len 00015358 (84 KiB): DRAM
MicroPy  Pyboard  I (1506) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
MicroPy  Pyboard  I (1513) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
MicroPy  Pyboard  I (1519) heap_init: At 4009AE3C len 000051C4 (20 KiB): IRAM
MicroPy  Pyboard  I (1526) cpu_start: Pro cpu start user code
MicroPy  Pyboard  I (1544) spi_flash: detected chip: generic
MicroPy  Pyboard  I (1545) spi_flash: flash io: dio
MicroPy  Pyboard  I (1545) cpu_start: Starting scheduler on PRO CPU.
MicroPy  Pyboard  I (0) cpu_start: Starting scheduler on APP CPU.
MicroPy  Pyboard  MicroPython v1.13-dirty on 2020-10-21; ESP32 module (spiram) with ESP32
MicroPy  Pyboard  Type "help()" for more information.
MicroPy  Failed to execute script: timeout waiting for first EOF reception
```

BradenM, 2020-10-23:
Interesting, seems micropython-stubber is having some issues with v1.13.

Just a guess, but I know the uasyncio module was completely rewritten for v1.13, so its possible it may now be causing some issues.

We could try excluding it from the stub list w/ a patch (see [here](https://github.com/Josverl/micropython-stubber/blob/master/patches/pycom.patch) for an example) and using it by passing the `patch` arg to the minify script method [here](https://github.com/BradenM/micropy-cli/blob/47be8e5de085bc1bffd328c4101c4c6be9c4b736/micropy/main.py#L102).

When I have some time on my hands, I will investigate this further. 

@Josverl, any thoughts?

Josverl, 2020-10-23:
There was a significant enum bug in mpy 1.13.0 where I ran into the same issue, and tried the path of excluding some modules.

Since then the bug has been found and solved in build 1.13.0-103.
In the latest version of stubber there is a warning for this, as well as inclusion of the new asyncio modules.

If you need the stubs for 1.13.0-103 or others, you can collect them on  https://github.com/Josverl/micropython-stubs

BradenM, 2021-05-17:
Closing due to being an upstream (micropython) issue (https://github.com/BradenM/micropy-cli/issues/167#issuecomment-714965627)

closed 2021-05-17 by BradenM