BradenM/micropy-cli #184: Error when initializing a project

bruno963852, 2020-11-12:
**Describe the bug**
When i run micropy init it crashes with the following error:
```Traceback (most recent call last):
  File "/home/brunomartins/.local/bin/micropy", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 73, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/brunomartins/.local/lib/python3.9/site-packages/micropy/cli.py", line 103, in init
    proj_path = project.create()
  File "/home/brunomartins/.local/lib/python3.9/site-packages/micropy/project/project.py", line 152, in create
    child.create()
  File "/home/brunomartins/.local/lib/python3.9/site-packages/micropy/project/modules/stubs.py", line 132, in create
    return self.load()
  File "/home/brunomartins/.local/lib/python3.9/site-packages/micropy/project/modules/stubs.py", line 120, in load
    self.config.upsert('stubs', {s.name: s.stub_version for s in self._stubs})
  File "/home/brunomartins/.local/lib/python3.9/site-packages/micropy/project/modules/stubs.py", line 120, in <dictcomp>
    self.config.upsert('stubs', {s.name: s.stub_version for s in self._stubs})
AttributeError: 'str' object has no attribute 'name'
```

**Logs**

```
[2020-11-11 16:25:17] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function PackagesModule.add_from_file at 0x7f03b740a040>, '_hook__add_from_file__path_None__dev_False')
[2020-11-11 16:25:17] DEBUG: hookproxy(add_package): Method added to proxy: (<function PackagesModule.add_package at 0x7f03b740a160>, '_hook__add_package__dev_False')
[2020-11-11 16:25:17] DEBUG: hookproxy(add_package): Method added to proxy: (<function DevPackagesModule.add_package at 0x7f03b740a5e0>, '_hook__add_package__dev_True')
[2020-11-11 16:25:17] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function DevPackagesModule.add_from_file at 0x7f03b740a700>, '_hook__add_from_file__path_None__dev_True')
[2020-11-11 16:25:17] DEBUG: hookproxy(stubs): Method added to proxy: (<function StubsModule.stubs at 0x7f03b73bcd30>, '_hook__stubs__')
[2020-11-11 16:25:17] DEBUG: hookproxy(add_stub): Method added to proxy: (<function StubsModule.add_stub at 0x7f03b73bd1f0>, '_hook__add_stub__')
[2020-11-11 16:25:17] DEBUG: micropy: MicroPy Loaded
[2020-11-11 16:25:17] DEBUG: micropy: Running first time setup...
[2020-11-11 16:25:17] DEBUG: micropy: Creating .micropy directory @ /home/brunomartins/.micropy
[2020-11-11 16:25:17] DEBUG: micropy: Starting new HTTPS connection (1): pypi.org:443
[2020-11-11 16:25:17] DEBUG: micropy: https://pypi.org:443 "GET /pypi/micropy-cli/json HTTP/1.1" 200 11648
[2020-11-11 16:25:17] INFO: micropy: Creating New Project
[2020-11-11 16:25:17] DEBUG: micropy: Using selector: EpollSelector
[2020-11-11 16:25:22] DEBUG: micropy: Starting new HTTPS connection (1): raw.githubusercontent.com:443
[2020-11-11 16:25:22] DEBUG: micropy: https://raw.githubusercontent.com:443 "GET /BradenM/micropy-stubs/master/source.json HTTP/1.1" 200 1439
[2020-11-11 16:25:22] ERROR: micropy: You don't have any stubs!
[2020-11-11 16:25:22] INFO: micropy: To add stubs to micropy, use micropy stubs add <STUB_NAME>
[2020-11-11 16:29:26] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function PackagesModule.add_from_file at 0x7f6fa9e06040>, '_hook__add_from_file__path_None__dev_False')
[2020-11-11 16:29:26] DEBUG: hookproxy(add_package): Method added to proxy: (<function PackagesModule.add_package at 0x7f6fa9e06160>, '_hook__add_package__dev_False')
[2020-11-11 16:29:26] DEBUG: hookproxy(add_package): Method added to proxy: (<function DevPackagesModule.add_package at 0x7f6fa9e065e0>, '_hook__add_package__dev_True')
[2020-11-11 16:29:26] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function DevPackagesModule.add_from_file at 0x7f6fa9e06700>, '_hook__add_from_file__path_None__dev_True')
[2020-11-11 16:29:26] DEBUG: hookproxy(stubs): Method added to proxy: (<function StubsModule.stubs at 0x7f6fa9db7d30>, '_hook__stubs__')
[2020-11-11 16:29:26] DEBUG: hookproxy(add_stub): Method added to proxy: (<function StubsModule.add_stub at 0x7f6fa9db81f0>, '_hook__add_stub__')
[2020-11-11 16:29:26] DEBUG: micropy: MicroPy Loaded
[2020-11-11 16:29:26] DEBUG: micropy: Starting new HTTPS connection (1): raw.githubusercontent.com:443
[2020-11-11 16:29:27] DEBUG: micropy: https://raw.githubusercontent.com:443 "GET /BradenM/micropy-stubs/master/source.json HTTP/1.1" 200 1439
[2020-11-11 16:29:27] DEBUG: bikegps: adding module: StubsModule
[2020-11-11 16:29:27] DEBUG: bikegps: adding module: PackagesModule
[2020-11-11 16:29:27] DEBUG: bikegps: adding module: DevPackagesModule
[2020-11-11 16:29:27] DEBUG: bikegps: adding module: TemplatesModule
[2020-11-11 16:29:27] INFO: micropy: Adding esp32-micropython-1.12.0 to stubs
[2020-11-11 16:29:27] INFO: stubs: Resolving stub...
[2020-11-11 16:29:27] DEBUG: micropy stubs: Stub Url: https://codeload.github.com/BradenM/micropy-stubs/legacy.tar.gz/pkg/esp32-micropython-1.12.0
[2020-11-11 16:29:27] DEBUG: micropy stubs: Starting new HTTPS connection (1): codeload.github.com:443
[2020-11-11 16:29:29] DEBUG: micropy stubs: https://codeload.github.com:443 "HEAD /BradenM/micropy-stubs/legacy.tar.gz/pkg/esp32-micropython-1.12.0 HTTP/1.1" 200 0
[... 1358 lines cut ...]
[2020-11-12 10:51:05] DEBUG: stubs: Validating: /home/brunomartins/.micropy/stubs/esp32-1.12.0
[2020-11-12 10:51:05] INFO: stubs: Detected Firmware: esp32 1.12.0
[2020-11-12 10:51:05] INFO: stubs: Firmware not found locally, attempting to install it...
[2020-11-12 10:51:05] INFO: stubs: Resolving stub...
[2020-11-12 10:51:05] DEBUG: micropy stubs: Stub Url: https://codeload.github.com/BradenM/micropy-stubs/legacy.tar.gz/pkg/esp32 1.12.0
[2020-11-12 10:51:05] DEBUG: micropy stubs: Starting new HTTPS connection (1): codeload.github.com:443
[2020-11-12 10:51:06] DEBUG: micropy stubs: https://codeload.github.com:443 "HEAD /BradenM/micropy-stubs/legacy.tar.gz/pkg/esp32%201.12.0 HTTP/1.1" 404 0
[2020-11-12 10:51:06] ERROR: stubs: Failed to resolve firmware!
[2020-11-12 10:51:06] INFO: stubs: esp32-1.12.0 added!
[2020-11-12 10:51:06] DEBUG: stubs: Loaded: esp32-1.12.0
[2020-11-12 10:51:08] DEBUG: bikegps: adding module: StubsModule
[2020-11-12 10:51:08] DEBUG: bikegps: adding module: PackagesModule
[2020-11-12 10:51:08] DEBUG: bikegps: adding module: DevPackagesModule
[2020-11-12 10:51:08] DEBUG: bikegps: adding module: TemplatesModule
[2020-11-12 10:51:08] INFO: bikegps: Initiating bikegps
[2020-11-12 10:51:08] DEBUG: bikegps: Generated Project Context: <micropy.config.config.Config object at 0x7f46894b4a00>
[2020-11-12 10:51:08] INFO: bikegps: Stubs: esp32-micropython-1.12.0
```

**Context (Environment)**

* OS: PopOS 20
* Micropy Version: 3.4.0 and 3.3.0
* Python Version: 3.8 and 3.9
* VSCode Version: 1.51.1

Fischchen, 2020-11-15:
I have the same Issue on Windows 10

atmuc, 2020-11-16:
I have the same issue. Is there any workaround for this bug?

Harvzor, 2020-11-16:
## Workaround

Running `micropy` shows the errors posted above.

I downgraded my install to a version which seemed to work:

```
> pip install micropy-cli==3.1.1
```

Verify install:

```
> micropy --version                          
micropy, version 3.1.1
```

Now I can run:

```
> micropy

MicroPy  Loading Project
MicroPy  ✔ Ready!

MicroPy  Update Available!
MicroPy  Version v3.4.0 is now available
MicroPy  You can update via: pip install --upgrade micropy-cli
```

BradenM, 2020-11-16:
Hey guys, sorry to see you guys are running into some issues.

I'm currently moving house, so I've been out of the loop for a bit, but as soon as I get situated I will see what I can do to help resolve this.

putyavka, 2020-11-17:
Unfortunately, @Harvzor workaround doesn't help in all cases. Error message is slitely different but basically the same

`PS D:\Dropbox\Electronics\RT-Thread\test2> micropy init     

MicroPy  Update Available!
MicroPy  Version v3.4.0 is now available
MicroPy  You can update via: pip install --upgrade micropy-cli

MicroPy  Creating New Project
? Project Name test2
? Choose any Templates to Generate  [VSCode Settings for Autocompletion/Intellisense]
? Which stubs would you like to use?  [esp32-micropython-1.12.0]

MicroPy  Initiating test2
Traceback (most recent call last):
  File "d:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Python36\Scripts\micropy.exe\__main__.py", line 7, in <module>
  File "d:\python36\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "d:\python36\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "d:\python36\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\python36\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\python36\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "d:\python36\lib\site-packages\click\decorators.py", line 73, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "d:\python36\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "d:\python36\lib\site-packages\micropy\cli.py", line 103, in init
    proj_relative = project.create()
  File "d:\python36\lib\site-packages\micropy\project\project.py", line 191, in create
    child.create()
  File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 128, in create
    f"Stubs: $[{' '.join(str(s) for s in self.stubs)}]")
  File "d:\python36\lib\site-packages\micropy\project\modules\modules.py", line 90, in wrapper
    return func(*args, **kwargs)
  File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 66, in stubs
    return self._resolve_subresource(self._stubs)
  File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 90, in _resolve_subresource
    self.parent.data_path))
  File "d:\python36\lib\site-packages\micropy\stubs\stubs.py", line 376, in resolve_subresource
    fware = stub.firmware
AttributeError: 'str' object has no attribute 'firmware'`

Windows 10
Python 3.6.4
micropy-cli 3.1.1
VSCode 1.51.1

For latest version of micropy-cli (3.4.0) the message is
`File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 132, in create
    return self.load()
  File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 120, in load
    self.config.upsert('stubs', {s.name: s.stub_version for s in self._stubs})
  File "d:\python36\lib\site-packages\micropy\project\modules\stubs.py", line 120, in <dictcomp>
    self.config.upsert('stubs', {s.name: s.stub_version for s in self._stubs})
AttributeError: 'str' object has no attribute 'stub_version'`

Harvzor, 2020-11-17:
I also gave up using `micropy-cli` after hitting the same error as above. 😬

tmbo, 2020-11-17:
the error is due to a regression in a dependency (questionary), if you upgrade to version `1.8.1` it should be fixed (e.g. `pip install questionary==1.8.1`)

BradenM, 2020-11-17:
@tmbo Oh wow, that was fast haha. Awesome, will update my PR to require `v1.8.1` instead of `<v1.8.0`. 

Thanks again.

closed 2020-11-17 by BradenM

BradenM, 2020-11-17:
Okay, thanks for the patience guys, you should now be able to run `pip install --upgrade micropy-cli` (`micropy --version` should report `v3.5.0` and no longer run into this issue.

PR #190 (2020-11-17): Constrain Questionary version to ^1.8.1
Constrains questionary to <1.8.0 due to a breaking change that was made that led to #184. 

See https://github.com/tmbo/questionary/issues/88 for more details.

- fix(pkg): setuptools editable installation issues
- fix(pkg): constrain questionary version to <1.8.0

**Update**

https://github.com/tmbo/questionary/issues/88 was resolved and released faster than I could merge this PR (CI issues :P), so have now updated the constraint to `^1.8.1` to include the fix.

Closes #184 

also changes/updates to workflows close #114
commit 2bec6c61 (2020-11-17): fix(pkg): constrain questionary version to <1.8.0

see https://github.com/tmbo/questionary/issues/88 for details

fixes #184