BradenM/micropy-cli #212: Cannot add stub (esp32-micropython-1.12.0)

alexbez, 2021-01-13:
When trying to add a stub, micropy crashes:

```
alexbez@T580:~$ micropy --version
micropy, **version 3.5.0**
alexbez@T580:~$ micropy stubs add esp32-micropython-1.12.0

MicroPy  Loading Project
Traceback (most recent call last):
  File "/usr/local/bin/micropy", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/decorators.py", line 73, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/micropy/cli.py", line 203, in add
    proj = mpy.project
  File "/usr/local/lib/python3.8/dist-packages/micropy/utils/decorators.py", line 20, in _lazy_property
    setattr(self, attr, fn(self))
  File "/usr/local/lib/python3.8/dist-packages/micropy/main.py", line 54, in project
    proj = self.resolve_project('.', verbose=self.verbose)
  File "/usr/local/lib/python3.8/dist-packages/micropy/main.py", line 77, in resolve_project
    proj.load()
  File "/usr/local/lib/python3.8/dist-packages/micropy/project/project.py", line 136, in load
    child.load(**kwargs)
  File "/usr/local/lib/python3.8/dist-packages/micropy/project/modules/templates.py", line 46, in load
    self.provider = self.get_provider(self.config.get('config'))
  File "/usr/local/lib/python3.8/dist-packages/micropy/project/modules/templates.py", line 42, in get_provider
    return TemplateProvider(templates, run_checks=self.run_checks, log=self.log)
  File "/usr/local/lib/python3.8/dist-packages/micropy/project/template.py", line 258, in __init__
    [self.TEMPLATES.get(t)[0] for t in templates]))
TypeError: 'NoneType' object is not iterable
alexbez@T580:~$ 
```
micropy log:
`[2021-01-13 18:24:18] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function PackagesModule.add_from_file at 0x7f8b392435e0>, '_hook__add_from_file__path_None__dev_False')
[2021-01-13 18:24:18] DEBUG: hookproxy(add_package): Method added to proxy: (<function PackagesModule.add_package at 0x7f8b39243700>, '_hook__add_package__dev_False')
[2021-01-13 18:24:18] DEBUG: hookproxy(add_package): Method added to proxy: (<function DevPackagesModule.add_package at 0x7f8b39243b80>, '_hook__add_package__dev_True')
[2021-01-13 18:24:18] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function DevPackagesModule.add_from_file at 0x7f8b39243ca0>, '_hook__add_from_file__path_None__dev_True')
[2021-01-13 18:24:18] DEBUG: hookproxy(stubs): Method added to proxy: (<function StubsModule.stubs at 0x7f8b391f2430>, '_hook__stubs__')
[2021-01-13 18:24:18] DEBUG: hookproxy(add_stub): Method added to proxy: (<function StubsModule.add_stub at 0x7f8b391f28b0>, '_hook__add_stub__')
[2021-01-13 18:24:57] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function PackagesModule.add_from_file at 0x7f4efb15d5e0>, '_hook__add_from_file__path_None__dev_False')
[2021-01-13 18:24:57] DEBUG: hookproxy(add_package): Method added to proxy: (<function PackagesModule.add_package at 0x7f4efb15d700>, '_hook__add_package__dev_False')
[2021-01-13 18:24:57] DEBUG: hookproxy(add_package): Method added to proxy: (<function DevPackagesModule.add_package at 0x7f4efb15db80>, '_hook__add_package__dev_True')
[2021-01-13 18:24:57] DEBUG: hookproxy(add_from_file): Method added to proxy: (<function DevPackagesModule.add_from_file at 0x7f4efb15dca0>, '_hook__add_from_file__path_None__dev_True')
[2021-01-13 18:24:57] DEBUG: hookproxy(stubs): Method added to proxy: (<function StubsModule.stubs at 0x7f4efb10c430>, '_hook__stubs__')
[2021-01-13 18:24:57] DEBUG: hookproxy(add_stub): Method added to proxy: (<function StubsModule.add_stub at 0x7f4efb10c8b0>, '_hook__add_stub__')
[2021-01-13 18:24:57] DEBUG: micropy: MicroPy Loaded
[2021-01-13 18:24:57] DEBUG: micropy: Running first time setup...
[2021-01-13 18:24:57] DEBUG: micropy: Creating .micropy directory @ /home/alexbez/.micropy
[2021-01-13 18:24:57] DEBUG: micropy: Starting new HTTPS connection (1): raw.githubusercontent.com:443
[2021-01-13 18:24:58] DEBUG: micropy: https://raw.githubusercontent.com:443 "GET /BradenM/micropy-stubs/master/source.json HTTP/1.1" 200 1479
[2021-01-13 18:24:58] DEBUG: micropy.config.config: loaded config values
[2021-01-13 18:24:58] DEBUG: alexbez: adding module: StubsModule
[2021-01-13 18:24:58] DEBUG: alexbez: adding module: PackagesModule
[2021-01-13 18:24:58] DEBUG: alexbez: adding module: DevPackagesModule
[2021-01-13 18:24:58] DEBUG: alexbez: adding module: TemplatesModule
[2021-01-13 18:24:58] INFO: micropy: Loading Project
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> {}
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> {}
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> {}
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> set()
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> {}
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [paths] -> []
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [stubs] -> {}
[2021-01-13 18:24:58] DEBUG: alexbez: loading requirements from: /home/alexbez/dev-requirements.txt
[2021-01-13 18:24:58] DEBUG: <packagedependencysource micropy-cli>: Starting new HTTPS connection (1): pypi.org:443
[2021-01-13 18:24:58] DEBUG: <packagedependencysource micropy-cli>: https://pypi.org:443 "GET /pypi/micropy-cli/json HTTP/1.1" 200 12268
[2021-01-13 18:24:58] DEBUG: alexbez: dumping to dev-requirements.txt
[2021-01-13 18:24:58] DEBUG: alexbez: dumping: ['micropy-cli\n']
[2021-01-13 18:24:58] DEBUG: micropy.config.config: set config value [paths] => [PosixPath('/home/alexbez/.micropy/alexbez1')]
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [dev-packages] -> ['micropy-cli']
[2021-01-13 18:24:58] DEBUG: micropy.config.config_source: creating new config file: /home/alexbez/.micropy/.cache
[2021-01-13 18:24:58] DEBUG: alexbez: loading requirements from: /home/alexbez/requirements.txt
[2021-01-13 18:24:58] DEBUG: alexbez: dumping to requirements.txt
[2021-01-13 18:24:58] DEBUG: alexbez: dumping: []
[2021-01-13 18:24:58] DEBUG: micropy.config.config: set config value [paths] => [PosixPath('/home/alexbez/.micropy/alexbez1')]
[2021-01-13 18:24:58] DEBUG: micropy.config.config: added config value [packages] -> []`

VoidDruid, 2021-01-17:
I'm having the exact same problem

BradenM, 2021-01-17:
Interesting... I will look into this further, but at first glance it appears micropy thinks it's currently in an active project from your home folder for some reason. 

Could you retry the command from a different directory and see if the you get the same results? 

Something like: `cd /tmp && micropy stubs add <stub>` 

Thanks

alexbez, 2021-01-18:
From /tmp directory it works fine.

```
alexbez@T580:~$ cd /tmp
alexbez@T580:/tmp$ micropy stubs add esp32-micropython-1.12.0

MicroPy  Adding esp32-micropython-1.12.0 to stubs

MicroPy  Resolving stub...
MicroPy  esp32-micropython-1.12.0: 100%|██████████████████████████████████████████████████████████████████████████████| [9.83k/9.83k @ 75.5MB/s]
MicroPy  Detected Firmware: micropython
MicroPy  Firmware not found locally, attempting to install it...

MicroPy  Resolving stub...
MicroPy  micropython: 100%|████████████████████████████████████████████████████████████████████████████████████████████| [43.4k/43.4k @ 176MB/s]
MicroPy  ✔ micropython firmware added!
MicroPy  ✔ esp32-micropython-1.12.0 added!
alexbez@T580:/tmp$ 
```

BradenM, 2021-05-17:
Closing due to staleness and unable to reproduce.
Feel free to let me know if this is still an issue.

closed 2021-05-17 by BradenM