BradenM/micropy-cli #217: Init fails on network drive (Win10)

daspiras, 2021-01-27:
OS: Win10
Micropy: 3.5
Python: 3.7
VSC: 1.52.1

When running `micropy init` on a mapped network drive from a NAS, it fails with:

`[2021-01-27 14:09:38] ERROR: test: Failed to Create Stub Links!
[2021-01-27 14:09:38] FileNotFoundError: [Errno 2] No such file or directory: 'Z:\\test\\.micropy\\BradenM-micropy-stubs-e1b8ce6\\info.json'`

Interestingly, micropy 
**Expected Behavior**

**Current Behavior**

**Steps to Reproduce**

1.
2.
3.

**Possible Solution**

**Logs**

**Context (Environment)**

* OS:
* Micropy Version:
* Python Version:
* VSCode Version:

BradenM, 2021-01-27:
Interesting. I wonder if this is some sort of permissions issue? 

As a workaround, you could try manually copying the stubs from your root micropy folder to where it's trying to create the links.

daspiras, 2021-01-28:
Yes. You are absolutely correct. I now see it was trying to create symlinks from `$HOME/.micropy` to the project folder. The fact it successfully created a .micropy folder on the network drive through me off.

As a test, I tried running `mklink` in a command prompt on network drive that contained my project folder and it was giving `access denied` errors, not allowing me to create a remote-> local link, so it is definitely a permissions problem on my end where micropy-cli can create files/directories on the network drive, but not the symlinks.

I tried various solutions to get symlinks working (just to have it functional generally) but chasing the problem through the QNAP-SMB-WIN10 ecosystem has not been fruitful and is not worth the effort.

I was trying to just use the network drive as a project folder so I could access the same project from both my desktop and laptop. I suspect that even if got symlinking working, it would still result in an issue because the link would point to whichever local $home for the device (desktop/laptop) I first ran `micropy init` on.

In the end, I just set up a git repository on the network drive and will just push/pull as needed.

FYI to anyone else in the same boat, manually copying the files seems to have worked as a work-around.

No bug, user error! Although perhaps adding a suggestion to check permissions may be helpful for folks in the future (or an option to copy files, not symlink). Thanks for the great project!

closed 2021-01-28 by daspiras

BradenM, 2021-01-30:
@daspiras Glad to hear you got it working. 

> No bug, user error! Although perhaps adding a suggestion to check permissions may be helpful for folks in the future (or an option to copy files, not symlink). Thanks for the great project!

I do agree it would be helpful to have more detailed output when this issue occurs. Also an option to fallback to copying is a good idea. 

The idea behind symlinking is to keep all your stubs centralized in one place, so that if you update/add any the changes reflect to any of your existing projects. That said, it would be nice to have a fallback for scenarios such as this. 

Thanks!