BradenM/micropy-cli #228: the need for something like `micropy deploy`

airtonix, 2021-02-21:
**Is your feature request related to a problem? Please describe.**

no... well yes. `micropy deploy` would be a great shortcut for micropython noobs.

Coming from the nodejs world, Currently I fee like micropy is just a combination of something like 
- `npm search|install`
- `yo generator:*`

Is it just trying to be a micropython packagemanager? or a good overall dev toolkit?

if the former then it should probably be split up into separate cli tools.
if the latter then we should also include a way for people to complete the workflow:

- setup
- install deps
- run simulator
- deploy to device

**Describe the solution you'd like**

something like `micropy deploy`.

at first it could just read various parameters required from `.micropy.json`.

Later on it could ask for target device type, etc to build the right settings in `.micropy.json`

the end effect would be micropython on my computer would end up on the connected device (in my case the m5paper).

**Describe alternatives you've considered**

bash scripts. 

**Additional context**

i'm a noob.

manuel2258, 2021-04-16:
I found micropy-cli because I was looking whether there is already a tool that uploads micropython source files to the microcontroller as that was the thing that always annoyed me about micropython.

If that is something that the maintainers would like to have or don't mind about I would love to implement it into micropy-cli.

Any suggestions about how this should work?

I was perhaps thinking that it might would be smart to track changes files and only upload changed once?

BradenM, 2021-11-22:
@manuel2258 If you are still interested in taking a stab at this, take a look at: #107 and let me know your thoughts.

Thanks