BradenM/micropy-cli #235: Why does it show a stub installed but then throw error

ilium007, 2021-03-02:
I am trying to test and installed the 'micropython' stub but when I then try to init a project it says no stubs installed

```
(micropython) micropython % micropy stubs search py                                                                                                                                            23:08:39

MicroPy  Searching Stub Repositories...

MicroPy  Results for py:
MicroPy  WiPy-pycom-1.18.2
MicroPy  esp32-micropython-1.10.0
MicroPy  esp32-micropython-1.11.0
MicroPy  esp32-micropython-1.12.0
MicroPy  esp32-micropython-1.9.4
MicroPy  esp32-pycopy-1.11.0
MicroPy  esp32-pycopy-2.11.0.1
MicroPy  esp32-pycopy-2.11.0.5
MicroPy  esp32-pycopy-3.0.0
MicroPy  esp8266-micropython-1.10.0
MicroPy  esp8266-micropython-1.11.0
MicroPy  esp8266-micropython-1.9.3
MicroPy  esp8266-micropython-1.9.4
MicroPy  esp8266-pycopy-1.11.0
MicroPy  esp8266-pycopy-2.11.0
MicroPy  esp8266-pycopy-2.11.0.1
MicroPy  ev3-pybricks-1.0.0
MicroPy  micropython (Installed)
MicroPy  pybricks
MicroPy  pycom
MicroPy  pycopy
(micropython) micropython % micropy init test1                                                                                                                                                 23:08:46

MicroPy  Creating New Project
? Choose any Templates to Generate done (4 selections)
MicroPy  You don't have any stubs!

MicroPy  To add stubs to micropy, use micropy stubs add <STUB_NAME>
(micropython) micropython %
```

ilium007, 2021-03-02:
Verified I have a stub in ~/.micropy/stubs

```
(micropython) micropython % micropy init test1                                                                                                                                                 23:14:41

MicroPy  Creating New Project
? Choose any Templates to Generate done (4 selections)
MicroPy  You don't have any stubs!

MicroPy  To add stubs to micropy, use micropy stubs add <STUB_NAME>
(micropython) micropython % ll ~/.micropy/stubs                                                                                                                                                23:14:57
drwxrwxr-x - xxxx  7 Sep  2020 BradenM-micropy-stubs-e1b8ce6
(micropython) micropython %
```

BradenM, 2021-03-25:
@ilium007 Sorry for my late reply. 

This is actually a design error. You installed the `micropython` stub, which is a "firmware" stub. Firmware stubs are more akin to dependencies rather than something micropy-cli uses to scaffold out a project. 

You need to install a "device" stub (such as `esp32-micropython-X.XX.X`).

Thanks for pointing this out.

closed 2021-05-17 by BradenM

BradenM, 2022-03-21:
Pinning this for visibility until it is fixed in v4.0.0