BradenM/micropy-cli #260: esp32-micropython-1.15.0 stub?

hsmptg, 2021-04-21:
Any chance of having a stub for the last micropython (1.15.0)?
The last one seems to be for version 1.12.0!
What is the impact of using a 1.12.0 stub in a 1.15 micropython?

```
micropy stubs search esp32

MicroPy  Searching Stub Repositories...

MicroPy  Results for esp32:
MicroPy  esp32-micropython-1.10.0
MicroPy  esp32-micropython-1.11.0
MicroPy  esp32-micropython-1.12.0 (Installed)
MicroPy  esp32-micropython-1.9.4
MicroPy  esp32-pycopy-1.11.0
MicroPy  esp32-pycopy-2.11.0.1
MicroPy  esp32-pycopy-2.11.0.5
MicroPy  esp32-pycopy-3.0.0
MicroPy  esp32_LoBo
MicroPy  esp32_LoBo-esp32_LoBo-3.2.24
```

Regards and thanks for this useful project.

hsmptg, 2021-04-22:
Hi
Meanwhile I found that I could create my own stubs as can be seen here:
https://micropy-cli.readthedocs.io/en/latest/base.html#creating-stubs

After creasting my esp32-micropython-1.15.0 stub, it seems to allow a working "lint".
For instance now if I use a feature introduced in v1.13 (machine-soft_reset()) it is recognised.

But I also found some minor problems.
For instance, if I write any method of uasyncio (for instance sleep_ms()) I get a warning:
Module 'uasyncio' has no 'sleep_ms' member

To avoid this I must write uasyncio.core.sleep_ms() instead of simply uasyncio.sleep_ms().
Note that micropython also accept this last one!
An alternative is to move in the stub folder the core.py file inside uasyncio folder to the main folder renaming it with uasyncio.py.

Regards

BradenM, 2021-05-17:
Hi @hsmptg,

I've finally gotten around to updating the stub repository and adding v1.15.0 (see: BradenM/micropy-stubs@eba97562a50502120749b626c0b61a8758d8363b).

Sorry for taking so long :grimacing: 

`micropy stubs add esp32-micropython-1.15.0` should work now, along with (I believe) properly working uasyncio stubs. 
Thanks.

closed 2021-05-17 by BradenM