BradenM/micropy-cli #275: raise ImportError(err)

cwhuasmile, 2021-05-27:
When I execute `micropy init newproject`
The following error message was encountered  
-------------------------------------------------------------
Traceback (most recent call last):
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\git\__init__.py", line 87, in <module>
    refresh()
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\git\__init__.py", line 76, in refresh
    if not Git.refresh(path=path):
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\git\cmd.py", line 309, in refresh
    raise ImportError(err)
ImportError: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\runpy.
py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\runpy.
py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\Scripts\mi
cropy.exe\__main__.py", line 5, in <module>
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\__init__.py", line 18, in <module>
    from micropy import data, lib, project, stubs, utils
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\project\__init__.py", line 5, in <module>
    from . import modules
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\project\modules\__init__.py", line 7, in <module>
    from .packages import DevPackagesModule, PackagesModule
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\project\modules\packages.py", line 12, in <module>
    from micropy.packages import (
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\packages\__init__.py", line 16, in <module>
    from .source_package import PackageDependencySource, VCSDependencySource
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\micropy\packages\source_package.py", line 9, in <module>
    from git import Repo
  File "c:\users\administrator\appdata\local\programs\python\python36\lib\site-p
ackages\git\__init__.py", line 89, in <module>
    raise ImportError('Failed to initialize: {0}'.format(exc)) from exc
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet
---------------------------------------------------------------
I changed the Python version from v3.7.9 to v3.6.8 and it still doesn't work  

* OS: windows 7 SP1
* Micropy Version: 3.6.0
* Python Version: 3.6.8
* VSCode Version: 1.56.2

BradenM, 2021-05-27:
Hi @cwhuasmile,

It seems you do not have Git installed (or it is not available on your path).

See: https://git-scm.com/download/win

Thanks.

BradenM, 2021-05-29:
Closing since the issue seems unrelated to micropy. 

Feel free to let me know otherwise. Thanks.

closed 2021-05-29 by BradenM

cwhuasmile, 2021-05-29:
@BradenM
Thank you very much. According to your tips, I have successfully solved my problem