BradenM/micropy-cli #279: Unmantained Pyminifier breaks on SetupTools >=58

robertoetcheverryr, 2021-09-15:
Hi!
I've started down the road of MicroPython and stumbled upon this magnificent project. It's just what I wanted, since I moved from the Arduino IDE to a complete IDE like PyCharm, to have the advanced features since not having code completion was really irking me.
I'm trying to make a complete guide on how to use lv_micropython from 0, including having a complete IDE that actually works, so this project was the last part of my journey (since I saw that some lv bindings still don't get updated automatically).

Now, when attempting to build the stubs from my already deployed firmware, I was asked to install the dependencies, and all but one was installed correctly. Pyminifier uses "use_2to3" which is currently failing in SetupTools 58 (I guess, due to Python2 being finally deprecated) and since it seems to have been left unmaintained I don't believe raising the issue over there would help. 
So, should an alternative to Pyminifier be found? should there be a requirement to keep SetupTools <58 for the time being?
Thanks in advance

Josverl, 2021-10-31:
I have updated micropython-stubber and `process.py` to use **python-minifier** which appears to work well.
It produces even more compact code, and I have needed to adjust a number of tests.

if you update to v1.4.1 and update the depedencies your existing code should work,
or rather than running the minification on demand you could just pick up the minified version from [`micropython-stubber/minified`](https://github.com/Josverl/micropython-stubber/tree/master/minified)

Josverl, 2021-11-01:
@robertoetcheverryr  ,
you may want to leave this open as @BradenM will still need to make a few changes to micropy-cli

robertoetcheverryr, 2021-11-08:
Reopened as per Josverl's comment

BradenM, 2021-11-22:
@robertoetcheverryr Thank you for the report. I've (finally :grimacing:) found the time to look into this and am dealing with it now.

@Josverl Thanks for finding a solution. Working on updating everything now.

commit 54cefd3a (2021-11-22): fix(stubber): replace pyminifer with python-minifer

pyminifier is no longer maintained

Closes: #279
PR #285 (2021-11-22): Update project dependencies and CI test workflow
Closes #279
commit a53ab4c6 (2021-11-22): fix(stubber): replace pyminifer with python-minifer

pyminifier is no longer maintained

Closes: #279
closed 2021-11-22 by BradenM