BradenM/micropy-cli #286: problem with micropy init =>ImportError Bad git executable.

hzhhgit, 2022-02-06:
hi
i did install micropy-cli using pip install  micropy-cli
and using pip list i can see this: micropy-cli         3.6.0
but on every command using micropy including micropy init i have follwoing error:
```
PS G:\isis-projects-2022\microPython\blink> micropy init
Traceback (most recent call last):
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\git\__init__.py", line 87, in <module>
    refresh()
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\git\__init__.py", line 76, in refresh
    if not Git.refresh(path=path):
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\git\cmd.py", line 340, in refresh
    raise ImportError(err)
ImportError: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the  
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:       
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\Scripts\micropy.exe\__main__.py", line 4, in <module>
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\__init__.py", line 18, in <module>
    from micropy import data, lib, project, stubs, utils
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\project\__init__.py", line 5, in <module>
    from . import modules
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\project\modules\__init__.py", line 7, in <module>
    from .packages import DevPackagesModule, PackagesModule
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\project\modules\packages.py", line 12, in <module>
    from micropy.packages import (
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\packages\__init__.py", line 16, in <module>
    from .source_package import PackageDependencySource, VCSDependencySource     
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\micropy\packages\source_package.py", line 9, in <module>
    from git import Repo
  File "C:\Users\Hamid\AppData\Roaming\Python\Python39\site-packages\git\__init__.py", line 89, in <module>
    raise ImportError('Failed to initialize: {0}'.format(exc)) from exc
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the  
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:       
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet
```
what is the problem and how can i solve that

hzhhgit, 2022-02-06:
my problem also solved using pip install micropy-cli==3.1.1

hzhhgit, 2022-02-06:
i alse downgarde to version 3.1.1 and that not working too

hzhhgit, 2022-02-06:
it seems my problem was with git and before any thing u should install git first and add it to system vaibales path

hzhhgit, 2022-02-06:
an other problem happend.
when i use micropy init and initialize project every thing work fine but when i close vscode and open it again autocomplete dosen't work

dalibor-sanzeru, 2022-02-09:
I had the same issue -> Intellisence didnt worked for python.
I have found language server is not configured when re-open second time VS Code after usage of micropy-cli:
Could you look in directory for this string?
    "python.languageServer"
  Change this to + save:
      "python.languageServer": "Pylance",
      
      
 File should be located at: .vscode/settings.json

hzhhgit, 2022-02-10:
> 

i don't konw how to appreciate you
your hint completey worked for me
as u said i just change `"python.languageServer": "None"` to "python.languageServer": "Pylance", and autocomplete worked again

closed 2022-02-10 by hhzhgit

dalibor-sanzeru, 2022-02-10:
@BradenM could you please take a look on this? 
Looks liike CLI is not storing python.languageServer after VS CODE restart.
Workround is in this thread.
Thanks.

BradenM, 2022-02-10:
> @BradenM could you please take a look on this? 
> Looks liike CLI is not storing python.languageServer after VS CODE restart.
> Workround is in this thread.
> Thanks.

Will see if I can take a look this weekend.