BradenM/micropy-cli #288: Running the cli command fails with trace

ShocKwav3, 2022-02-20:
Hi,

I am fairly new to micropython development and so far it's great. I was trying to setup the `micropy-cli` in order to get my vscode stop complaining about those errors. Unfortunately, I can't run the any command `micropy-cli`. They fails with the following error trace:

````

Traceback (most recent call last):
  File "/Users/ferojother/Library/Python/3.10/bin/micropy", line 5, in <module>
    from micropy.cli import cli
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/micropy/__init__.py", line 18, in <module>
    from micropy import data, lib, project, stubs, utils
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/micropy/project/__init__.py", line 5, in <module>
    from . import modules
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/micropy/project/modules/__init__.py", line 9, in <module>
    from .templates import TemplatesModule
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/micropy/project/modules/templates.py", line 7, in <module>
    from micropy.project.template import TemplateProvider
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/micropy/project/template.py", line 11, in <module>
    from jinja2 import Environment, FileSystemLoader
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/jinja2/__init__.py", line 12, in <module>
    from .environment import Environment
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/jinja2/environment.py", line 25, in <module>
    from .defaults import BLOCK_END_STRING
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/jinja2/defaults.py", line 3, in <module>
    from .filters import FILTERS as DEFAULT_FILTERS  # noqa: F401
  File "/Users/ferojother/Library/Python/3.10/lib/python/site-packages/jinja2/filters.py", line 13, in <module>
    from markupsafe import soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/Users/ferojother/Library/Python/3.10/lib/python/site-packages/markupsafe/__init__.py)
````

Any pointers on how I can fix this?

Thanks.

BradenM, 2022-02-21:
Hi @ShocKwav3, thanks for the report.
Looks like micropy got bit by https://github.com/pallets/jinja/issues/1585.

Will have a fix out soon. Thanks.

lukich48, 2022-02-27:
@ShocKwav3 you can downgrade markupsafe to version 2.0.1 while fix is preparing

pip install --upgrade markupsafe==2.0.1

closed 2022-03-14 by BradenM

BradenM, 2022-03-15:
@ShocKwav3 @lukich48 

This should be fixed in the latest release candidate (v4.0.0.rc.1).
Would be awesome if any of you guys could give it a try and let me know of any issues :smile: 

Do note that (as of now) the v4 candidate does not support creating stubs on Windows when using python 3.10 (anything >=3.9 is supported). This is due to rshell depending on pyreadline, which breaks on windows w/ py 3.10.

That said, I am working on a big improvement to how micropy talks with pyboards with support for using either rhell or https://github.com/Carglglz/upydevice as backends. Aiming to have this out prior to a final v4 release.

PR #289 (2022-03-14): Drop Python 3.6, Upgrade+Pin jinja2 to v3.0.3
Closes #288

Got bit by https://github.com/pallets/jinja/issues/1585

Also updates pytest to ^v7.0 to fix tests crashing with py10+win, see:
pyreadline/pyreadline#65 and pytest-dev/pytest#8733