BradenM/micropy-cli #299: Missing modules from generated stubs

golyalpha, 2022-04-23:
I'm running a version of Micropython that includes lvgl and it's bindings, including some display drivers. Unfortunately, when I run the stubber via micropy, it doesn't even find these additional modules - even though they are present, and can be imported and used just fine.

BradenM, 2022-04-23:
@golyalpha,

Currently, micropy uses a pretty dated version micropython-stubber (https://github.com/Josverl/micropython-stubber/tree/aefb2c9ea6bd2ff6ce5daff6f494115b0dd2e472) for generating stubs. As of now, it utilizes a [predefined list of target modules](https://github.com/Josverl/micropython-stubber/blob/aefb2c9ea6bd2ff6ce5daff6f494115b0dd2e472/board/createstubs.py#L557-L692) to check for & stub which only includes a subset of lvgl modules. (This is as of the latest v4.0.0-rc.2 release).

With #294 completed, I have plans to take advantage of the numerous improvements+features that @Josverl has introduced in the past several months along with greatly expanding this side of micropy-cli's capabilities. 

Included in this is the planned deprecation of the micropy-stubs repository in favor of searching/remotely installing from the far better maintained https://github.com/Josverl/micropython-stubs 

For now, I would recommend checking https://github.com/Josverl/micropython-stubs/tree/main/stubs for the stubs you are looking for.
You can either use these directly as documented in that repository or add to micropy via a local path.

If you are still unable to find what you are looking for there, take a look at the documentation available @ https://github.com/Josverl/micropython-stubber for information on how to generate your stubs using the latest stubber.

Josverl, 2022-04-25:
@golyalpha 

Please have a look at the available lvgl stubs : 
https://github.com/Josverl/micropython-stubs/tree/main/stubs/lvgl-v8_1_0_dev-esp32

that were created following a request and discussion.
https://github.com/Josverl/micropython-stubs/discussions/554

closed 2023-05-19 by BradenM