BradenM/micropy-cli #312: pymakr config setup depricated

justus-saul, 2022-11-02:
* OS: Manjaro Linux rolling
* Micropy Version: v4.0.0-rc.2
* Python Version: 3.10
* VSCode Version: 1.7.30
* pymakr: 2.22.5

Hi,
I just setup my first project with the help of Micropy Cli on VSCode.
It seems the config Micropy is creating for pymakr in its currently available version from the extensions is deprecated.
Of all options set by Micropy only `safe_boot_on_upload` and `py_ignore` are still valid.

The proposed workflow sadly isn't working anymore and you currently have to upload each file manually.
I guess the only way would be downgrading pymakr manually? (Or finding another workaround)

BradenM, 2022-11-05:
Thanks for bringing this to my attention.

I will look into adding support for pymakr v2 config over this weekend, hopefully. :+1:

jshamg, 2023-03-03:
> Thanks for bringing this to my attention.
> 
> I will look into adding support for pymakr v2 config over this weekend, hopefully. 👍

Any news on this? :)

BradenM, 2023-03-05:
@justus-saul @jshamg 

Have not quite gotten around to it, sorry.

I have not used pymakr in quite a while, would either of you be willing to share an example of what you'd be looking for in a pymakr v2 template?

Thanks.

jshamg, 2023-03-05:
Hi @BradenM ,
no worries, thanks for the reply! :)
So my biggest Problem is, that pymakr apparently doesn't support the sync of a single folder anymore. I don't think they really meant to do that but there is just no setting for it anymore. there is another option apart from sync_folder called dist_dir but if i set it I get the warning "Property dist_dir is not allowed."... Apart from that this is a complete config I would be looking for :
```
{
  "name": "SomeProjectName",
  "dist_dir": "src",
  "ctrl_c_on_connect": false,
  "safe_boot_on_upload": false,
  "reboot_after_upload": true,
  "py_ignore": [
    "pymakr.conf",
    ".vscode",
    ".gitignore",
    ".git",
    "project.pymakr",
    "env",
    "venv",
    ".python-version",
    ".micropy/",
    "micropy.json"
  ],
  "dev": {
    "simulateDeepSleep": false,
    "uploadOnDevStart": "outOfSync",
    "onUpdate": "restartScript"
  }
}
```
I've also made a [feature request](https://github.com/pycom/pymakr-vsc/issues/295) at the pymakr-vsc repo for the folder sync problem. So maybe waiting for that would be smart :)

Thanks!

Josverl, 2023-03-05:
Pymakr was under development by pycom, which went article 11 (bankrupt)

Still use the preview version, as I've found nothing better for vscode on windows,..

justus-saul, 2023-03-06:
Sorry, for the late response.
For me there is no incentive to upgrade pymakr to v2. I just need an easy way for uploading the project to the device.
Downgrading it to v1 was fine for me, so just reflecting that in the docs would be fine, but in the long run it isn't that nice to depend on something that is not getting support anymore.

BradenM, 2023-03-06:
Seems it would be best to simply offer both PyMakr v1 and v2 template options  as an immediate solution. 

However, based on the @Josverl and @justus-saul insights along with the lackluster sync support pointed out by @jshamg in pymakr v2, it seems the need for #228 has just grown and should be prioritized. 

I have already laid the foundation for a built-in library sync / code sync with the PyDevice abstractions introduced in #294 and (with some needed improvements to micropys configuration) aim to make this a reality in the near future.

Please let me know if any of you guys agree with this sentiment -- and if so perhaps we should open a discussion to fully layout what an ideal implementation would look like in micropy / the scope of features wanted.

jshamg, 2023-03-07:
Yes and no. On the one hand I would say It could potially be a nice enhancement to have micropy-cli deploy to devices in general. The ability to sync it with the right configuration directly to the device is really nice. On the other hand i'd like to have a integration in vs-code and pymakr v2 is doing it pretty well with organising py-boards, naming them and syncing them. For me personally i rather do it through vs-code than via terminal, but I'm not limited to pymakr if there are any other solutions...

justus-saul, 2023-03-07:
I first need to be honest in the regard that my micro python project is currently on my pile of shame since the start of 2023, so everything is kind of based on older memory, but I guess we can also speak about more general concepts here.

I would say consistency would be the key for me and that would be having everything I need within micropy-cli. Being dependent on a plugin/tool is okay as long it you don't have an additional workflow for it. As with pymakr this is currently not the case as we use its interface, config, etc.. I am all in for an integrated solution. This step would also make micropy-cli less reliant on VCS for an important part of the workflow of a project.

Josverl, 2023-04-04:
Pymakr 2 code is available from github, and several forks also exist.

Most connection issues I see are down to multiple instances of vscode all trying to grab the same devices. Tweaking the auto connect settings solve that.

With mpremote and mip there are quite a few ways to deploy packages, although requirements / pyproject.toml files are still not working for micropython...