BradenM/micropy-cli #338: Installed latest micropy. After micropy init I get error

guineau, 2022-11-12:
macOS Ventura on M1 MacBook.

micropy, version 4.0.0rc2
Python 3.9.13
VSCode Version: 1.73.1

Installed micropy and ran:

$ micropy stubs install esp32-micropython-1.10.0
$ micropy init 

Both of which worked fine.

However when I run micropy now I always get the following stack dump.

(upy) guineau:~/Projects/ESP/ATOM  % micropy stubs list                                 

MicroPy  Installed Stubs:
MicroPy  Total: 1

MicroPy  Micropython:
MicroPy  esp32-micropython-1.10.0
Traceback (most recent call last):
  File "/Users/guineau/miniconda3/envs/upy/bin/micropy", line 8, in <module>
    sys.exit(cli())
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/decorators.py", line 73, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/cli.py", line 256, in list
    proj = mpy.project
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/utils/decorators.py", line 20, in _lazy_property
    setattr(self, attr, fn(self))
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/main.py", line 56, in project
    proj = self.resolve_project(".", verbose=self.verbose)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/main.py", line 79, in resolve_project
    proj.load()
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/project/project.py", line 133, in load
    child.load(**kwargs)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/project/modules/templates.py", line 47, in load
    self.provider = self.get_provider(self.config.get("config"))
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/project/modules/templates.py", line 43, in get_provider
    return TemplateProvider(templates, run_checks=self.run_checks, log=self.log)
  File "/Users/guineau/miniconda3/envs/upy/lib/python3.9/site-packages/micropy/project/template.py", line 257, in __init__
    chain.from_iterable([self.TEMPLATES.get(t)[0] for t in templates])
TypeError: 'NoneType' object is not iterable
(upy) guineau:~/Projects/ESP/ATOM  %

BradenM, 2022-11-22:
Hi @guineau, thanks for the report.

After created a few dummy projects with different variations, I was able to recreate your error.

It seems to happen when no templates are selected during the `micropy init` process. This currently results in an (incorrect) micropy.json file that lacks a `config` key entirely, instead of a `config` key with an empty object as its value.

Can you confirm that you did not select any templates?

If so, a workaround for now would be to manually add `"config": {}` to your `micropy.json` file. This should resolve the issue until I can implement a proper fix in the next release.

```patch
--- /tmp/bad/micropy.json	2022-11-21 18:02:25.417349824 -0600
+++ /tmp/good/micropy.json	2022-11-21 18:02:37.960437026 -0600
@@ -6,5 +6,7 @@
     "dev-packages": {
         "micropy-cli": "*"
     },
-    "packages": {}
+    "packages": {},
+    "config": {}
 }
```

If it was not your intention to not select any templates, you may remove the `.micropy/` and `micropy.json` files and rerun the `micropy init` command.

guineau, 2022-11-22:
Hi Braden,

I am new to micropy (and MicroPython!) so I probably just set things up incorrectly. 

Ok, I just deleted the project and started again - and I totally missed the part about "press space to select" when picking templates. Much better now that I selected something :)

Sorry for confusion and thank you for a great utility!

You can close this bug report (or should I?)

john

BradenM, 2022-11-23:
No worries! 

Best to just leave this issue open for now, as it is a legitimate bug for micropy to crash in the scenario that no templates were selected.

Glad that you have a working setup now though, and best of luck on your project(s)!

commit ae8f5d8a (2023-03-27): feat(app): extract main cli logic out to command/option callbacks.

Additionally resolve some known issues and make minor improvements.
Fixed #338
commit d59449ad (2023-03-27): feat(app): extract main cli logic out to command/option callbacks.

Additionally resolve some known issues and make minor improvements.
Fixed #338
commit e3f6401f (2023-03-27): feat(app): extract main cli logic out to command/option callbacks.

Additionally resolve some known issues and make minor improvements.
Fixed #338