BradenM/micropy-cli #342: MicroPy-CLI Crashing during stubs create:

seth-c-stenzel, 2022-11-15:
**Describe the bug**
When running near the end of the create process I'm getting an `StopIteration` error.
It almost seems to me like everything is successful, just that the upload process is failing, and I'm unsure if this means the stubbing process completed correctly and just an upload failed, or what, the error seems ambiguous to me.

Stubs folder is empty.

```
File "/home/sstenzel/.local/lib/python3.10/site-packages/micropy/stubs/stubs.py", line 328, in from_stubber
    mod_file = next(_path.rglob("modules.json"))
StopIteration
```

**Expected Behavior**
Should finish running and not crash.

**Current Behavior**
Seems to be failing near the end of the process.

**Steps to Reproduce**
1. Install Linux
2. `pip install micropy-cli`
3. `pip install micropy-cli[create_stubs]`
4. `micropy stubs create /dev/ttyUSB0`
[micropy.log](https://github.com/BradenM/micropy-cli/files/10014486/micropy.log)

**Logs**
[micropy.log](https://github.com/BradenM/micropy-cli/files/10014486/micropy.log)

* OS: Windows 10, Ubuntu 22.04LTS
* Micropy Version: MicroPython v1.19.1-645-g451ded8d7
* Python Version: 3.10.5, 3.10.6, 3.11.0

BradenM, 2022-11-19:
Hi @seth-c-stenzel, thanks for the report. 

> Micropy Version: MicroPython v1.19.1-645-g451ded8d7

While this is useful to know, what version of `micropy-cli` were you using when this happened? Thanks.

JayPalm, 2023-01-05:
I am having the exact same issue, using micropy-cli v4 (installed via pip, and using pyenv):
```log
Traceback (most recent call last):
  File "/Users/<USER_HOME>/.pyenv/versions/mpy/bin/micropy", line 8, in <module>
    sys.exit(cli())
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/micropy/cli.py", line 276, in create
    return mpy.create_stubs(port, verbose=verbose)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/micropy/main.py", line 132, in create_stubs
    stub_path = self.stubs.from_stubber(stub_path, out_dir)
  File "/Users/<USER_HOME>/.pyenv/versions/3.10.8/envs/mpy/lib/python3.10/site-packages/micropy/stubs/stubs.py", line 328, in from_stubber
    mod_file = next(_path.rglob("modules.json"))
StopIteration
```

What's somewhat strange is that I was previously able to generate stubs for the Lobo micropython fork, but I've since moved to building my own firmware from the latest official commit.

BradenM, 2023-05-18:
@seth-c-stenzel @JayPalm Hi, I suspect this issue should be resolved with a more recent release of micropy. Please let me know if otherwise. Thanks!

closed 2023-05-18 by BradenM

ThatOtherAndrew, 2024-05-12:
Latest version of micropy-cli (4.2.2), I'm having the same StopIteration error. Full logs provided below:

```
❯ micropy stubs create COM3

MicroPy  Connecting to Pyboard @ COM3
MicroPy  ✔ Connected!
2024-05-12 01:55:20.373 | DEBUG    | stubber.minify:minify_script:253 - Original length : 26905
2024-05-12 01:55:20.375 | INFO     | stubber.minify:minify_script:254 - Minified length : 11631
2024-05-12 01:55:20.375 | INFO     | stubber.minify:minify_script:255 - Reduced by      : 15274
2024-05-12 01:55:52.302 | DEBUG    | stubber.minify:cross_compile:328 - mpy-cross compiled to    : mpy_cross_9dxwiuiu.mpy
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Writing /createstubs.mpy: 100%|████████████████████████████████████████████| [10.4k/10.4k @ 2.34kB/s]
MicroPy  Pyboard  Port: rp2
MicroPy  Pyboard  Board: Pimoroni_Badger2040W_2MB_with_RP2040
MicroPy  Pyboard  Clean/remove files in folder: /stubs/micropython-v1_21_0-dirty-rp2-Pimoroni_Badger2040W_2MB_with_RP2040
MicroPy  Pyboard  Stub module: _boot_fat                 to file: _boot_fat.py
                 mem:138880
MicroPy  Pyboard  Stub module: _onewire                  to file: _onewire.py
                 mem:140160
MicroPy  Pyboard  Stub module: _rp2                      to file: _rp2.py
                 mem:139776
MicroPy  Pyboard  Stub module: _thread                   to file: _thread.py
                 mem:139648
MicroPy  Pyboard  Stub module: adcfft                    to file: adcfft.py
                 mem:139504
MicroPy  Pyboard  Stub module: aioble.__init__           to file: __init__.py
                 mem:133424
MicroPy  Pyboard  Stub module: aioble.central            to file: central.py
                 mem:133696
MicroPy  Pyboard  Stub module: aioble.client             to file: client.py
                 mem:131184
MicroPy  Pyboard  Stub module: aioble.core               to file: core.py
                 mem:131456
MicroPy  Pyboard  Stub module: aioble.device             to file: device.py
                 mem:131280
MicroPy  Pyboard  Stub module: aioble.l2cap              to file: l2cap.py
                 mem:128304
MicroPy  Pyboard  Stub module: aioble.peripheral         to file: peripheral.py
                 mem:128192
MicroPy  Pyboard  Stub module: aioble.security           to file: security.py
                 mem:127472
MicroPy  Pyboard  Stub module: aioble.server             to file: server.py
                 mem:127488
MicroPy  Pyboard  Stub module: array                     to file: array.py
                 mem:127344
MicroPy  Pyboard  Stub module: binascii                  to file: binascii.py
                 mem:127200
MicroPy  Pyboard  Stub module: bluetooth                 to file: bluetooth.py
                 mem:127056
MicroPy  Pyboard  Stub module: breakout_as7262           to file: breakout_as7262.py
                 mem:126832
MicroPy  Pyboard  Stub module: breakout_bh1745           to file: breakout_bh1745.py
                 mem:126672
MicroPy  Pyboard  Stub module: breakout_bme280           to file: breakout_bme280.py
                 mem:126512
MicroPy  Pyboard  Stub module: breakout_bme68x           to file: breakout_bme68x.py
                 mem:126352
MicroPy  Pyboard  Stub module: breakout_bmp280           to file: breakout_bmp280.py
                 mem:126192
MicroPy  Pyboard  Stub module: breakout_dotmatrix        to file: breakout_dotmatrix.py
                 mem:126032
MicroPy  Pyboard  Stub module: breakout_encoder          to file: breakout_encoder.py
                 mem:125872
MicroPy  Pyboard  Stub module: breakout_icp10125         to file: breakout_icp10125.py
                 mem:125712
MicroPy  Pyboard  Stub module: breakout_ioexpander       to file: breakout_ioexpander.py
                 mem:125536
MicroPy  Pyboard  Stub module: breakout_ltr559           to file: breakout_ltr559.py
                 mem:125392
[... 355 lines cut ...]
│ │        stub_path = WindowsPath('C:/Users/Andrew/AppData/Local/Temp/tmpo1mrr16m/stubs')       │ │
│ │           tmpdir = 'C:\\Users\\Andrew\\AppData\\Local\\Temp\\tmpo1mrr16m'                    │ │
│ │          variant = <CreateStubsVariant.BASE: 'base'>                                         │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                                  │
│ C:\Users\Andrew\AppData\Local\Programs\Python\Python311\Lib\site-packages\micropy\stubs\stubs.py │
│ :355 in from_stubber                                                                             │
│                                                                                                  │
│   352 │   │   """                                                                                │
│   353 │   │   _path = Path(path).resolve()                                                       │
│   354 │   │   dest = Path(dest).resolve()                                                        │
│ ❱ 355 │   │   mod_file = next(_path.rglob("modules.json"))                                       │
│   356 │   │   path = mod_file.parent                                                             │
│   357 │   │   mod_data = json.load(mod_file.open())                                              │
│   358 │   │   dev_fware = mod_data["firmware"]                                                   │
[... cut ...]
│ │  dest = WindowsPath('C:/Users/Andrew/AppData/Local/Temp/tmpo1mrr16m')       │                  │
│ │  path = WindowsPath('C:/Users/Andrew/AppData/Local/Temp/tmpo1mrr16m/stubs') │                  │
│ │  self = <micropy.stubs.stubs.StubManager object at 0x000002336F86C810>      │                  │
│ ╰─────────────────────────────────────────────────────────────────────────────╯                  │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
StopIteration
```