BradenM/micropy-cli #345: "int", "str", "bytes", etc. not defined

armeniopinto, 2022-11-30:
Hello,

I've just started using `micropy-cli` and it has been a great experience: smooth and clean. The only problem that I have is that Pylance is complaining that it can't find definitions for common Python functions - none of them is being found! Do you know what I might be missing here? Thank you!

armeniopinto, 2022-12-01:
Hello again,

I installed `micropy-cli` on another system, in a completely clean VSC installation, and it has the same problem.

BradenM, 2022-12-01:
Hi @armeniopinto 

Thanks for the reports. Micropy currently does not have templating support for pylance built in, unfortunately. (Maybe be wip, see #306). 

See #152 for some workarounds to make it work for now though.

armeniopinto, 2022-12-02:
Hello, makes sense, thank you!

BradenM, 2023-01-30:
Closed in #379

closed 2023-01-30 by BradenM