BradenM/micropy-cli #368: Feature for Vim autocomplete

ozkaya84, 2023-01-27:
Hi,

Does anyone use this from Nvim ? 
I wonder whether there is a possibility for autocomplete in nvim...

BradenM, 2023-01-29:
Hi @ozkaya84, 
Please see:
https://github.com/BradenM/micropy-cli/discussions/302#discussioncomment-3730836

Thanks

closed 2023-01-29 by BradenM

ozkaya84, 2023-01-30:
Thank you Branden.
I did manage to source ./micropy libs to vim editor.
Steps :
- Creating Virtual Env (with venv python3 module)
- Adding  "<Virtual Home>/.micropy/stubs/esp32-micropython-1.18.0/stubs" path  (or what you have) folder to custom.pth file to virtual env python lib folder as:  
`lib/python3.10/site-packages/custom.pth 
`and inside the file:

`<your full path>/.micropy/stubs/BradenM-micropy-stubs-ebac1d3/stubs
`
and thats all:
[image]