BradenM/micropy-cli #440: stubs create process get stuck but won't give any errors

mttdr, 2023-04-08:
Hello, I'm trying to create stubs from my Pico Lipo board (a clone of Raspberry Pi Pico, with added functionality) because I want support for auto-completion of libraries like `picographics` that are added by Pimoroni and not present in the standard rp2 stubs.

I run `micropy stubs create /dev/ttyACM0` the process will start, proceed for a bit and then always get stuck at the same step when copying stubs (and a CPU thread will be at 100% until I cancel the command).

See below for the complete log:
```
$ micropy stubs create /dev/ttyACM0 

MicroPy  Update Available!
MicroPy  Version v4.2.0b2 is now available
MicroPy  You can update via: pip install --upgrade micropy-cli

MicroPy  Connecting to Pyboard @ /dev/ttyACM0
MicroPy  ✔ Connected!
Original length : 23816
Minified length : 8338
Reduced by      : 15478 
MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Writing /createstubs.py: 100%|███████████████████████████████████████████████████████| [8.14k/8.14k @ 2.28kB/s]
MicroPy  Pyboard  Clean/remove files in folder: /stubs/micropython-rp2-1_19_1
MicroPy  Pyboard  Clean/remove files in folder: /stubs/micropython-rp2-1_19_1/uasyncio
MicroPy  Pyboard  Stub module: _thread              to file: /stubs/micropython-rp2-1_19_1/_thread.py                mem:189408
MicroPy  Pyboard  Stub module: ak8963               to file: /stubs/micropython-rp2-1_19_1/ak8963.py                 mem:189312
MicroPy  Pyboard  Stub module: apa102               to file: /stubs/micropython-rp2-1_19_1/apa102.py                 mem:189312
MicroPy  Pyboard  Stub module: apa106               to file: /stubs/micropython-rp2-1_19_1/apa106.py                 mem:189312
MicroPy  Pyboard  Stub module: array                to file: /stubs/micropython-rp2-1_19_1/array.py                  mem:189312
MicroPy  Pyboard  Stub module: binascii             to file: /stubs/micropython-rp2-1_19_1/binascii.py               mem:189216
MicroPy  Pyboard  Stub module: btree                to file: /stubs/micropython-rp2-1_19_1/btree.py                  mem:189120
MicroPy  Pyboard  Stub module: builtins             to file: /stubs/micropython-rp2-1_19_1/builtins.py               mem:188992
MicroPy  Pyboard  Stub module: cmath                to file: /stubs/micropython-rp2-1_19_1/cmath.py                  mem:188896
MicroPy  Pyboard  Stub module: collections          to file: /stubs/micropython-rp2-1_19_1/collections.py            mem:188784
MicroPy  Pyboard  Stub module: crypto               to file: /stubs/micropython-rp2-1_19_1/crypto.py                 mem:188672
MicroPy  Pyboard  Stub module: curl                 to file: /stubs/micropython-rp2-1_19_1/curl.py                   mem:188672
MicroPy  Pyboard  Stub module: dht                  to file: /stubs/micropython-rp2-1_19_1/dht.py                    mem:188672
MicroPy  Pyboard  Stub module: display              to file: /stubs/micropython-rp2-1_19_1/display.py                mem:188576
MicroPy  Pyboard  Stub module: display_driver_utils to file: /stubs/micropython-rp2-1_19_1/display_driver_utils.py   mem:188560
MicroPy  Pyboard  Stub module: ds18x20              to file: /stubs/micropython-rp2-1_19_1/ds18x20.py                mem:188560
MicroPy  Pyboard  Stub module: errno                to file: /stubs/micropython-rp2-1_19_1/errno.py                  mem:188464
MicroPy  Pyboard  Stub module: esp                  to file: /stubs/micropython-rp2-1_19_1/esp.py                    mem:188336
MicroPy  Pyboard  Stub module: esp32                to file: /stubs/micropython-rp2-1_19_1/esp32.py                  mem:188336
MicroPy  Pyboard  Stub module: espidf               to file: /stubs/micropython-rp2-1_19_1/espidf.py                 mem:188336
MicroPy  Pyboard  Stub module: flashbdev            to file: /stubs/micropython-rp2-1_19_1/flashbdev.py              mem:188336
MicroPy  Pyboard  Stub module: framebuf             to file: /stubs/micropython-rp2-1_19_1/framebuf.py               mem:188336
MicroPy  Pyboard  Stub module: freesans20           to file: /stubs/micropython-rp2-1_19_1/freesans20.py             mem:188240
MicroPy  Pyboard  Stub module: fs_driver            to file: /stubs/micropython-rp2-1_19_1/fs_driver.py              mem:188240
MicroPy  Pyboard  Stub module: functools            to file: /stubs/micropython-rp2-1_19_1/functools.py              mem:188224
MicroPy  Pyboard  Stub module: gc                   to file: /stubs/micropython-rp2-1_19_1/gc.py                     mem:188096
MicroPy  Pyboard  Stub module: gsm                  to file: /stubs/micropython-rp2-1_19_1/gsm.py                    mem:188000
MicroPy  Pyboard  Stub module: hashlib              to file: /stubs/micropython-rp2-1_19_1/hashlib.py                mem:188000
MicroPy  Pyboard  Stub module: heapq                to file: /stubs/micropython-rp2-1_19_1/heapq.py                  mem:187904
MicroPy  Pyboard  Stub module: ili9XXX              to file: /stubs/micropython-rp2-1_19_1/ili9XXX.py                mem:187808
MicroPy  Pyboard  Stub module: imagetools           to file: /stubs/micropython-rp2-1_19_1/imagetools.py             mem:187808
MicroPy  Pyboard  Stub module: inisetup             to file: /stubs/micropython-rp2-1_19_1/inisetup.py               mem:187808
MicroPy  Pyboard  Stub module: io                   to file: /stubs/micropython-rp2-1_19_1/io.py                     mem:187808
MicroPy  Pyboard  Stub module: json                 to file: /stubs/micropython-rp2-1_19_1/json.py                   mem:187712
MicroPy  Pyboard  Stub module: lcd160cr             to file: /stubs/micropython-rp2-1_19_1/lcd160cr.py               mem:187616
MicroPy  Pyboard  Stub module: lodepng              to file: /stubs/micropython-rp2-1_19_1/lodepng.py                mem:187616
MicroPy  Pyboard  Stub module: logging              to file: /stubs/micropython-rp2-1_19_1/logging.py                mem:187616
MicroPy  Pyboard  Stub module: lv_colors            to file: /stubs/micropython-rp2-1_19_1/lv_colors.py              mem:187616
MicroPy  Pyboard  Stub module: lv_utils             to file: /stubs/micropython-rp2-1_19_1/lv_utils.py               mem:187616
MicroPy  Pyboard  Stub module: lvgl                 to file: /stubs/micropython-rp2-1_19_1/lvgl.py                   mem:187616
MicroPy  Pyboard  Stub module: lwip                 to file: /stubs/micropython-rp2-1_19_1/lwip.py                   mem:187616
MicroPy  Pyboard  Stub module: machine              to file: /stubs/micropython-rp2-1_19_1/machine.py                mem:187616
MicroPy  Pyboard  Stub module: math                 to file: /stubs/micropython-rp2-1_19_1/math.py                   mem:187520
MicroPy  Pyboard  Stub module: microWebSocket       to file: /stubs/micropython-rp2-1_19_1/microWebSocket.py         mem:187360
MicroPy  Pyboard  Stub module: microWebSrv          to file: /stubs/micropython-rp2-1_19_1/microWebSrv.py            mem:187232
MicroPy  Pyboard  Stub module: microWebTemplate     to file: /stubs/micropython-rp2-1_19_1/microWebTemplate.py       mem:187216
MicroPy  Pyboard  Stub module: micropython          to file: /stubs/micropython-rp2-1_19_1/micropython.py            mem:187232
MicroPy  Pyboard  Stub module: mpu6500              to file: /stubs/micropython-rp2-1_19_1/mpu6500.py                mem:187136
MicroPy  Pyboard  Stub module: mpu9250              to file: /stubs/micropython-rp2-1_19_1/mpu9250.py                mem:187136
[... 93 lines cut ...]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/io.py: 100%|█████████████████████████████████████████| [657/657 @ 467B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/json.py: 100%|███████████████████████████████████████| [236/236 @ 391B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/machine.py: 100%|████████████████████████████████| [4.11k/4.11k @ 599B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/math.py: 100%|███████████████████████████████████████| [910/910 @ 503B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/micropython.py: 100%|████████████████████████████████| [370/370 @ 368B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/modules.json: 100%|██████████████████████████████| [2.12k/2.12k @ 568B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/neopixel.py: 100%|███████████████████████████████████| [290/290 @ 289B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/onewire.py: 100%|████████████████████████████████████| [518/518 @ 368B/s]
[... cut ...]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/uasyncio/funcs.py: 100%|█████████████████████████████| [245/245 @ 406B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/uasyncio/lock.py: 100%|██████████████████████████████| [286/286 @ 285B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/uasyncio/stream.py: 100%|████████████████████████| [1.39k/1.39k @ 544B/s]
MicroPy  Pyboard  Reading /stubs/micropython-rp2-1_19_1/ubinascii.py: 100%|██████████████████████████████████| [264/264 @ 263B/s]

```

BradenM, 2023-04-08:
Hi @mttdr

Thanks for opening up an issue.

I would recommend upgrading to the latest beta `v4.2.0b2` via `pip install --upgrade micropy-cli` / `pipx upgrade micropy-cli` (depending on how you installed).

There has been some much needed improvements since the latest stable in regards to stubs creation in that beta, so I expect you may see some very different results.

I would also take a look at #400 for more information (and why the beta has not quite been pushed out yet.) 

If you are able to give that a try, would appreciate any updates regarding how it worked out for you. Thanks.

mttdr, 2023-04-08:
Unfortunately it still gets stuck when copying stubs, this time at cmath.py:
```
$ micropy stubs create /dev/ttyACM0 

MicroPy  Connecting to Pyboard @ /dev/ttyACM0
MicroPy  ✔ Connected!
2023-04-08 11:28:55.032 | DEBUG    | stubber.minify:minify_script:251 - Original length : 25279
2023-04-08 11:28:55.032 | INFO     | stubber.minify:minify_script:252 - Minified length : 10476
2023-04-08 11:28:55.032 | INFO     | stubber.minify:minify_script:253 - Reduced by      : 14803 
MicroPy  Executing stubber on pyboard...

[removed for brevity]

MicroPy  Pyboard  Created stubs for 100 modules on board micropython-v1.19.1-rp2
MicroPy  Pyboard  Path: /stubs/micropython-v1_19_1-rp2
MicroPy  ✔ Done!
MicroPy  Copying stubs...
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/_boot_fat.py: 100%|█████████████████████████████████| [173/173 @ 287B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/_onewire.py: 100%|██████████████████████████████████| [310/310 @ 309B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/_rp2.py: 100%|██████████████████████████████████████| [988/988 @ 546B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/_thread.py: 100%|███████████████████████████████████| [426/426 @ 424B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/_uasyncio.py: 100%|█████████████████████████████████| [367/367 @ 365B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/adcfft.py: 100%|████████████████████████████████████| [272/272 @ 271B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/array.py: 100%|█████████████████████████████████████| [269/269 @ 268B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/binascii.py: 100%|██████████████████████████████████| [291/291 @ 290B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_as7262.py: 100%|███████████████████████████| [826/826 @ 457B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_bh1745.py: 100%|███████████████████████████| [509/509 @ 507B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_bme280.py: 100%|███████████████████████████| [640/640 @ 455B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_bme68x.py: 100%|███████████████████████████| [736/736 @ 523B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_bmp280.py: 100%|███████████████████████████| [642/642 @ 456B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_dotmatrix.py: 100%|████████████████████████| [468/468 @ 466B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_encoder.py: 100%|██████████████████████████| [576/576 @ 410B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_icp10125.py: 100%|█████████████████████████| [386/386 @ 384B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_ioexpander.py: 100%|███████████████████| [1.02k/1.02k @ 474B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_ltr559.py: 100%|███████████████████████████| [759/759 @ 540B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_matrix11x7.py: 100%|███████████████████████| [340/340 @ 338B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_mics6814.py: 100%|█████████████████████████| [787/787 @ 435B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_msa301.py: 100%|███████████████████████| [1.25k/1.25k @ 578B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_paa5100.py: 100%|██████████████████████████| [544/544 @ 387B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_pmw3901.py: 100%|██████████████████████████| [544/544 @ 387B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_potentiometer.py: 100%|████████████████████| [546/546 @ 388B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_rgbmatrix5x5.py: 100%|█████████████████████| [342/342 @ 341B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_rtc.py: 100%|██████████████████████████| [2.13k/2.13k @ 570B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_scd41.py: 100%|████████████████████████████| [405/405 @ 403B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_sgp30.py: 100%|████████████████████████████| [572/572 @ 407B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_trackball.py: 100%|████████████████████████| [586/586 @ 417B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/breakout_vl53l5cx.py: 100%|█████████████████████████| [846/846 @ 468B/s]
MicroPy  Pyboard  Reading /stubs/micropython-v1_19_1-rp2/cmath.py: 100%|█████████████████████████████████████| [392/392 @ 390B/s]
```

Looking at the issue you link maybe it's worth mentioning that the board I'm using has 4mb of flash

BradenM, 2023-05-17:
@mttdr Following up on this -- I suspect this issue has been resolved in the most recent stable release of micropy. Have you had a chance to try it?

BradenM, 2023-05-18:
I am going to go ahead and close this as I suspect it as been resolved. Please let me know if otherwise. Thanks

closed 2023-05-19 by BradenM

mttdr, 2023-05-19:
excuse me about the late reply.

Yes I did manage to create stubs correctly. although I'm not sure if it was thanks to the update or to the new board I'm using (the first one was giving me other headaches and thankfully they sent me a new one)

BradenM, 2023-05-19:
@mttdr No problem, and happy to hear you've had some success.