BradenM/micropy-cli #517: vscode missing imports after creating project

maxi07, 2023-06-23:
**Describe the bug**
Hey there! I am trying to configure a new project for an esp32. With your tool i have downloaded the stubs and created a new project with the micropy cli. The project is created, but VS code still complains about failing to resolve the `import machine`.

[image]

**Steps to Reproduce**

1. Create an empty vs code project
2. Add esp32 stubs
3. Init a new micropy project

The stubs then are added to the project and linked in the `settings.json`:
```json
{
    // Loaded Stubs:  micropython-esp32-stubs  micropython-stdlib-stubs 
    
    "python.languageServer": "Pylance",
    "python.analysis.autoSearchPath": true,
    "python.autoComplete.extraPaths": [".micropy\\micropython_esp32_stubs-1.19.1.post12", ".micropy\\micropython-stdlib-stubs-0.9.0\\stdlib", ".micropy\\BradenM-micropy-stubs-4f5a52a\\frozen", ".micropy\\esptest"],
    "python.analysis.diagnosticSeverityOverrides": { "reportMissingModuleSource": "none" },
    "python.analysis.typeCheckingMode": "basic",
    "python.autoComplete.typeshedPaths":  [".micropy\\micropython_esp32_stubs-1.19.1.post12", ".micropy\\micropython-stdlib-stubs-0.9.0\\stdlib", ".micropy\\BradenM-micropy-stubs-4f5a52a\\frozen", ".micropy\\esptest", "typings"],
    "python.analysis.typeshedPaths":  [".micropy\\micropython_esp32_stubs-1.19.1.post12", ".micropy\\micropython-stdlib-stubs-0.9.0\\stdlib", ".micropy\\BradenM-micropy-stubs-4f5a52a\\frozen", ".micropy\\esptest", "typings"],
    
    

    "python.linting.enabled": true,
    "python.linting.pylintEnabled": true
}
```

Please see my demo video here:

https://github.com/BradenM/micropy-cli/assets/7480270/dcba4164-dc32-4997-9552-61a0dbab1e77

But even after a vscode restart, the import cannot be resolved.
**Logs**
[micropy.log](https://github.com/BradenM/micropy-cli/files/11853060/micropy.log)

**Context (Environment)**

* OS: Win 11 Version 10.0.22621.1848
* Micropy Version: 4.2.1
* Python Version: 3.11.3
* VSCode Version: 1.79.2

BradenM, 2023-06-24:
@maxi07 Thanks for the report.

> But even after a vscode restart, the import cannot be resolved. I tracked your discussion https://github.com/BradenM/micropy-cli/discussions/471, but this also does not work.

In the discussion you linked, I mentioned that opening the project in a vscode workspace causes this issue, but opening it directly works fine. In the video you attached, it appears you are opening in a workspace. Are you implying that you have also tried running `code .` from the newly created project directory and that still does not work?

maxi07, 2023-06-24:
no, when i cd into the folder and open with `code .`, i still receive the same result. 

https://github.com/BradenM/micropy-cli/assets/7480270/3e5e64c0-7c2f-4d9c-a665-a45cdf3d94c5

BradenM, 2023-06-25:
@maxi07,

When you ran `micropy init esptest` inside the `sources\repos\esptest` directory, it created a new directory: `sources\repos\esptest\esptest\`

Therefore, you are opening the parent directory (`sources\repos\esptest`) and not the generated project directory. You should be running `code esptest` assuming you are within the `sources\repos\esptest` directory.

If it is your intention to create the project within `sources\repos\esptest`, then you should run `micropy init .`

maxi07, 2023-06-25:
thank you, this solved the issue! perhaps something to mention in your documentation for getting started?

closed 2023-06-25 by maxi07