BradenM/micropy-cli #521: Problem after updating to the latest version

elatobi, 2023-07-18:
I updatet micropy-cli to the latest version and i can't get it work again. After the first install i head a lots of trouble to get it work so i never  updated it. But now i did it and can't remember how i managed it to work. I'm using a MacOS 13.4.1 VSCode 1.65.2.
when I run the init command i get the following error message 
`Tobi@Tobis-MBP-16 Test_1 % micropy init
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/micropy", line 5, in <module>
    from micropy.cli import cli
ModuleNotFoundError: No module named 'micropy.cli'`

I'm very thankful for any help 😁

BradenM, 2023-07-19:
Hi @elatobi,

This seems to be an issue with your python environment or related.
That error is indeed correct that `micropy.cli` is not a module.

It was, however, a module in prior versions... so either something went wrong with your installation or your shell may be referencing an older installation binary (may be a PATH issue). Without some more details about your setup and how you actually went about installing it, that's about all I can say.

I would recommend looking into pipx:
https://pypa.github.io/pipx/

It's a tool that allows you to install python executables into isolated python virtual environments.

BradenM, 2023-07-22:
:+1: Closing this assuming you figured your issue out.
You are welcome to reopen if needed.

closed 2023-07-22 by BradenM