BradenM/micropy-cli #525: getting various errors

Mrqozy, 2023-08-24:
**Describe the bug**
im getting many errors such as "Expected no arguments to "Pin" constructor"

**Expected Behavior**

**Current Behavior**

**Steps to Reproduce**

1.
2.
3.

**Possible Solution**

**Logs**

**Context (Environment)**

* OS:
* Micropy Version:
* Python Version:
* VSCode Version:

Mrqozy, 2023-08-25:
> @Mrqozy,
> 
> Hi. This gives very little to go off and I cannot help without more details. Please add logs as well as more context around what stubs you are using, what device you are using, your configuration, etc.

im using micropython-esp32-stubs and almost every other stub give "print" is not defined" and "range" is not defined" error.

im using esp32

pymakr.conf
{
    "address": "192.168.4.1",
    "username": "micro",
    "password": "python",
    "sync_folder": "src",
    "open_on_start": true,
    "safe_boot_on_upload": false,
    "py_ignore": [
        "pymakr.conf",
        ".vscode",
        ".gitignore",
        ".git",
        "project.pymakr",
        "env",
        "venv",
        ".python-version",
        ".micropy/",
        "micropy.json"
    ],
    "fast_upload": false
}

[image]

Mrqozy, 2023-08-25:
print and range are recognized after commenting out this line

UchihaYuki, 2023-09-01:
> im getting many errors such as "Expected no arguments to "Pin" constructor"

Have you solved your problem? I have encountered the same thing.

UchihaYuki, 2023-09-01:
Instead of using `esp32-micropython-1.15.0` as the stubs of my project, I switch to  the following:
```json
"stubs": {
    "micropython-esp32-stubs": "1.19.1.post13",
    "micropython-stdlib-stubs": "0.9.0"
}
```
Everything works now. But I still have other problems.

Mrqozy, 2023-09-01:
> > im getting many errors such as "Expected no arguments to "Pin" constructor"
> 
> Have you solved your problem? I have encountered the same thing.

Like I mentioned comment these lines in the settings.json under the folder .vscode

For pwm error you can just type next to the error line #type: ignore

This will tell pylance to ignore the error.

It's not a perfect solution but it's a start, until now the creators of this resp. didn't reply again even after I provided the info they asked for.

Mrqozy, 2023-09-01:
> > im getting many errors such as "Expected no arguments to "Pin" constructor"
> 
> Have you solved your problem? I have encountered the same thing.

[image]
this is the fix, frequency is not positional argument but a keyword one

Mrqozy, 2023-09-01:
sorry for the mistake i just had to use a keyword argument for the frequency not a positional argument

closed 2023-09-01 by Mrqozy