BradenM/micropy-cli #533: E   ModuleNotFoundError: No module named 'requirements'

kianmeng, 2023-09-02:
**Describe the bug**
Cannot run test.

**Expected Behavior**
`pytest` should run without errors.

**Current Behavior**
```
$ make test
pytest
ImportError while loading conftest '/tmp/micropy-cli/tests/conftest.py'.
tests/conftest.py:8: in <module>
    import micropy
micropy/__init__.py:18: in <module>
    from micropy.main import MicroPy
micropy/main.py:9: in <module>
    from micropy import data, utils
micropy/utils/__init__.py:10: in <module>
    from .helpers import (
micropy/utils/helpers.py:24: in <module>
    import requirements
E   ModuleNotFoundError: No module named 'requirements'
```

**Steps to Reproduce**

1. `git clone https://github.com/BradenM/micropy-cli.git`
2. `cd micropy-cli`
3. `poetry shell`
4. `poetry install`
5. `make test`

**Possible Solution**

**Logs**

**Context (Environment)**

* OS: Ubuntu 23.04
* Micropy Version: `master` branch
* Python Version: 3.11.4
* VSCode Version: 1.81.1

kianmeng, 2023-09-06:
@BradenM Not working here?

```sh
$ poetry install -E test -E lint -E docs
Installing dependencies from lock file

Extra [test] is not specified.

$ poetry --version
Poetry (version 1.3.2)
```

BradenM, 2023-09-06:
@kianmeng Apologies, that was a mistake on my part.

What I meant was `--with` to specify the optional dependency groups:
`poetry install --with test --with lint --with docs`

See:
https://python-poetry.org/docs/master/managing-dependencies/#dependency-groups

Also, I would recommend using a more recent poetry version (>=1.4.1)

closed 2023-09-06 by kianmeng