BradenM/micropy-cli #575: ModuleNotFoundError: No module named 'imp'

ResRipper, 2024-01-17:
**Describe the bug**

Python 3.12.1 is currently unsupported, trying to install `micropy-cli` will result in an error due to missing the `imp` module.

**Expected Behavior**

Install successfully with no error.

**Current Behavior**

Install will produce an error that says:
```
ModuleNotFoundError: No module named 'imp'
```
Specifically, it is caused by the `pathtools`, which was validated by trying to install it separately:

```
╰─ pip install pathtools==0.1.2                                                                      ─╯
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Collecting pathtools==0.1.2
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/e7/7f/470d6fcdf23f9f3518f6b0b76be9df16dcc8630ad409947f8be2eb0ed13a/pathtools-0.1.2.tar.gz (11 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [20 lines of output]
      Traceback (most recent call last):
        File "C:\Users\xxx\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\xxx\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\xxx\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-_i0yj9q1\overlay\Lib\site-packages\setuptools\build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-_i0yj9q1\overlay\Lib\site-packages\setuptools\build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-_i0yj9q1\overlay\Lib\site-packages\setuptools\build_meta.py", line 480, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-_i0yj9q1\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 25, in <module>
      ModuleNotFoundError: No module named 'imp'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

According to the [Python documentation](https://docs.python.org/3.11/library/imp.html), `imp` was deprecated in Python 3.4, and removed in 3.12.

**Steps to Reproduce**

1. Install Python 3.12.1
2. Install `micropy-cli` with `pip install --upgrade micropy-cli` or `pip install --upgrade --pre micropy-cli`

**Logs**

```
[... 66 lines cut ...]
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-s64_ngzu\overlay\Lib\site-packages\setuptools\build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-s64_ngzu\overlay\Lib\site-packages\setuptools\build_meta.py", line 480, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "C:\Users\xxx\AppData\Local\Temp\pip-build-env-s64_ngzu\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 25, in <module>
      ModuleNotFoundError: No module named 'imp'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

**Context (Environment)**

* OS: Windows 11
* Micropy Version: 4.2.2
* Python Version: 3.12.1

whinee, 2024-02-11:
Use python 3.11. `imp` is deprecated and is turned into `importlib` in 3.12.

jamesbramich, 2024-10-08:
Can I just check there is still no plan to fix this?

bebbo, 2024-12-02:
> Use python 3.11. `imp` is deprecated and is turned into `importlib` in 3.12.

replace `import imp` with `import importlib as imp`

KoStard, 2025-07-17:
`uv tool install micropy-cli --python 3.11` this worked for me.